rs2k/raceflight #4: Ex.Bus (Jeti) support för Raceflight?

kajakmannen, 2015-11-23:
Ex.Bus (Jeti) support för Raceflight? 
Any code wizard that can make it happen? :D pretty please with suger on top.

Here is the documentation for Ex.bus protocol which maybe makes it easier to determine if it is possible or not.

http://www.jetimodel.com/en/show-file/642/

Thanks in advance for someone to reply on this issue, Kajaken

rs2k, 2015-11-25:
It shouldn't be too difficult to implement, but I don't have time to do it myself right now. There are other things that need to happen first.

blckmn, 2016-01-07:
I'll take a look on ebay see if i can't get a cheap second hand one of these to give this a go.

blckmn, 2016-01-07:
Crikey. Those Jeti units are expensive! I don't think I'll be getting one anytime soon. :P

hydra, 2016-01-18:
@kajakmannen Best to request for cleanflight support, then @rs2k can merge it after it's done and tested by the cleanflight team including @ProDrone who looks after the failsafe support.

hydra, 2016-01-18:
Ahh, I see you already requested it here: https://github.com/cleanflight/cleanflight/issues/1513

You can probably close this issue.

blckmn, 2016-04-10:
https://github.com/cleanflight/cleanflight/pull/1709 has been brought into raceflight here https://github.com/rs2k/raceflight/pull/143

If it can be tested @kajakmannen it can be merged.

@marv-t if you are happy also.

kajakmannen, 2016-04-11:
So did I understand you right that ex.bus is implemented in raceflight now or??  Not so used to the routines here on got hub :) I would love to test it. Exbus should make it possible for telemetry also right?

marv-t, 2016-04-11:
At the moment the following telemetry data are transmitted:
voltage, current, capacity and altitude.

On GPS I'm still working (waiting for GPS module).

Small changes are still in the pipeline, but basically no effect on the current functions.

@kajakmannen 
Please connect rx and tx from the fc with a 2k7 resistor. Connect the receiver on the rx pin.
-> have fun

proggod, 2016-06-19:
Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod