rs2k/raceflight #11: how to use the bin file

xfce, 2015-12-19:
use st-jlink download at 0x08000000?

rs2k, 2015-12-19:
For the revo and sparky2 the correct address is 0x08020000.
For the revo nano it's 0x08010000

You can upload the bin using the OP or TL bootloader and their GCS as well.

xfce, 2015-12-20:
ok, thanks. I have flash the nano use OP's GCS-15.05.2,
it seems OK, alarm and status LED can flash, and on the PC list two STM32 Virtual ComPort in FS Mode, can't find the driver. Where can I find the driver?
another thing I use  CLEANFLIGHT configurator 1.1.0, does it OK? thanks!

blackmanj, 2015-12-20:
http://www.st.com/web/en/catalog/tools/PF257938

Once you run the first install - navigate to the install directory, and run the .exe for your system (64bit or 32bit). The first install is not the driver - just an extractor for the driver.

xfce, 2015-12-20:
OK , Thanks very much, I have fixed it. now all seems OK

closed 2015-12-20 by xfce