rs2k/raceflight #42: Won't arm too many tasks running - scheduler

blckmn, 2016-01-08:
if (isCalibrating() || (averageWaitingTasks100 > **100**)) {
            warningLedFlash();
            DISABLE_ARMING_FLAG(OK_TO_ARM);
        } else {
            if (ARMING_FLAG(OK_TO_ARM)) {
                warningLedDisable();
            } else {
                warningLedFlash();
            }
        }

blckmn, 2016-01-08:
Consistently sitting at about 400+ starts at above 700... but comes down and stalls at 410-420

rs2k, 2016-01-10:
I'll take a look at this when I get back in town.

blckmn, 2016-01-15:
given the calculation for the averageWaitingTasks100 includes a multiple of 100 ... then the condition > 100 will imply only 1 waiting task. Perhaps it is meant to be > 1000? to allow for up to 10 tasks to be running in the background? I have mine set at 500 and it arms without issue.

rs2k, 2016-01-15:
I'll set it to 1000 and see what happens

rs2k, 2016-01-16:
Seems to be okay at 1000.

rs2k, 2016-01-18:
No problems and excellent flight results at 1000 on both 4KHz and 8KHz motor output.

blckmn, 2016-01-18:
Works great for me too :) thanks @rs2k

closed 2016-01-18 by blckmn