rs2k/raceflight #43: OneShot125: Updating motors after every control loop

blckmn, 2016-01-10:
With control loop at 250us, there is not enough delay before next motor signal at full throttle.

This means most ESCs won't calibrate. Confirmed on scope, and confirmed problem with BLheli and Simonk ESCs. @MJ666 advises similar symptoms on AlienFlightF4.

ESCs work fine with OneShot disabled.

rs2k, 2016-01-10:
I haven't had any problems calibrating oneshot at 4KHz. I'm only calibrating to a max throttle of 1850 to avoid this problems. Multishot will eventually solve this problem.

blckmn, 2016-01-11:
I'll play with fastPWM and see if it works instead.

MJ666, 2016-01-11:
Not an option here yet since we use 12A an 18A Kiss ESC's. We will give calibrating with 1850 an try soon.

blckmn, 2016-01-11:
Just added some code to test ... essentially allowing the use of motor_pwm_rate for OneShot125.

Setting it to 800 (using CLI) and the Lumenier ESCs calibrate ok. I can then set it to 2000 and the ESCs still run fine (but with an update every 500us) :)

MJ666, 2016-01-15:
We just tested this fix with the kiss ESC's. This working well. Thanks.

blckmn, 2016-01-15:
Awesome! :) I'll close this one then.

closed 2016-01-15 by blckmn