rs2k/raceflight #46: CLI: Garbage returned with SET

blckmn, 2016-01-11:
I've noticed that SET does not return a complete set of details, and often has garbage at points.

@rs2k and @MJ666 do you notice similar issues? could be the VCP.

rs2k, 2016-01-11:
I can't verify this since I'm still in Vegas, but I looked at the code and have an idea why. I had to introduce a delay between lines with the CLI dump command and I can do the same with the Set command. I don't know if this is a problem due to CPU or VCP speed and the task scheduler or the VCP driver itself yet, but the delay works as a good band-aid for now.

MJ666, 2016-01-11:
I have seen this sometimes with some of the builds during the last days. Sometimes the CLI dumps are not correct and parts are missing.  It also lookseems to happen more frequently if the fc is running for quite some time.

blckmn, 2016-01-11:
I'll check out the delay you added for the dump, and see if it works with the SET. Thx.

blckmn, 2016-01-12:
Adding the delay fixed the issue.

I also noticed the gyro_sampling settings lookup table was not complete - though further investigation it is not the sampling rate - but the LPF - not sure what was the intention there.

blckmn, 2016-01-13:
Ok - this problem seems to be related to the VCP only sending a byte at a time. It seems it is only an issue in the CLI of cleanflight (due to the javascript handling of the inbound data). Normal terminal is fine.

There are some patches in the cleanflight repo awaiting merging from last year. I will take a look and see if they resolve the issue.

I always dislike adding delays ;)

rs2k, 2016-01-18:
I agree about delays. They're messy bandaids at best.

hydra, 2016-01-18:
Agreed, don't spend to much time fixing the CLI or related issues as the intention is to remove it in time.

Please see https://github.com/cleanflight/cleanflight/issues/922

blckmn, 2016-01-18:
Ok - thanks @hydra. I won't spend any time on the CLI - but buffering improvements will still be beneficial in general for the comms around downloading the blackbox flash etc. The VCP feels like a shocker at the moment (exhibited in the CLI and BB flash downloads).

closed 2016-01-18 by blckmn

hydra, 2016-01-20:
the buffering and stuff is already merged here, been using it for a while. will be merging after CF 1.12