rs2k/raceflight #54: target.c - thoughts

blckmn, 2016-01-15:
What are the thoughts on adding initialisations such as MPU interrupt etc in to a specific target file (in the target directory) to clean up the initialisation.c etc?

MJ666, 2016-01-16:
I would move everything target specific in the target.h file. May be standardizing this file more into specific sections. To spread the configuration into multiple files is may be not very useful. Definitely the taget specific parts should be removed from common code. 

In my view we also should work in a way wich allow backporting the feature the CleanFlight and BetaFlight will be possible. This may not involve the flight code itself so much but the hardware related parts like the drivers etc. Ideally common changes should be made in Cleanflight and merged back to BetaFlight -> RaceFlight. But how fast the development is progressing this is may be a dream.

blckmn, 2016-01-17:
Only a thought :)

We could implement target specific functions like:
const extiConfig_t *selectMPUIntExtiConfig(void)
or even clean up pwm_mapping.c (it's quite complex)

MJ666, 2016-01-17:
I thing the concept used for the serial, I2C and SPI driver works quite well. Have generic function wich parametrized with definitions from the target.h file.

henn1001, 2016-01-17:
Something like this is quite easy to do for the exti handler
https://github.com/henn1001/cleanflight/commit/ac60d5a7ed499d7b7bd9caf496fbea7451d1f535
(not finished yet - there are still some issues)

blckmn, 2016-01-17:
Hi @henn1001 , yeah that's the approach taken for most of the driver files too.

MJ666, 2016-01-17:
This was exactly what i meant. Only for the Naze it would get an bit tricky (AllienFlight F3 V2 will have the same problem). May be this can be solved like i did for the LED driver in this pending PR:

https://github.com/cleanflight/cleanflight/pull/1638
https://github.com/MJ666/cleanflight/blob/AlienFlightF3_V2/src/main/target/ALIENWIIF3/target.h#L27-L42

Still not very elegant but it works. May be someone has an better idea?

blckmn, 2016-01-17:
Should the different board revisions (where they are breaking changes) be separate targets? So naze32_rev4 as a target?

MJ666, 2016-01-17:
I think it is more simple for users to deal with an single firmware for different board revisions as long as possible? Would be quite complicated for them to always need to understand what revision they have.

For AlienFlight F3 V2 we need to reassign some pins to free them up for SPI so we implemented an hardware detection and handle the differences in the firmware. Finally a single firmware for all boards.

I my view it will not be not very practical to have separate firmware for Rev3 - Rev6 for the Naze in example?

henn1001, 2016-01-17:
Yes the naze hardware revision detection is quite ugly as it´s done more or less on startup. So #defines won´t work here.  Different targets for that would be a cleaner implementation but not sure if this is wanted by hydra?

Only other option would be to create a seperate board.h and board.c file where all the init is done.  I think then the timer /pwm configuration could be easier implemented on a per target base.

MJ666, 2016-01-17:
I think if I remember well @ledvinap already started some project like this a wile ago. Unfortunately the milestones not working at the moment for Cleanflight so I did not find it. I think this was one part @hydra wants to get solved before adding F4 support to Cleanflight.

henn1001, 2016-01-17:
Yeah he has a lot of improvements in line.

hydra, 2016-01-17:
Yes, the EXTI and IO stuff is coming first.  This should make things alot easier and simpler.  The main motivation behind the cleanups are, so you are aware:
1) runtime allocation of resources so users can choose which pins they want to use using a UI.
2) removal of as much target conditional and configuration code as possible.  They're essentially all very very similar due to the CPU resources.
3) removal of the horrid pwm_mapping.c code.
4) remove the need for the target specific code in timer.c.

I will be looking at https://github.com/cleanflight/cleanflight/pull/1421 after Cleanflight 1.12.0 is released.  It's top of my list.

rs2k, 2016-01-18:
@hydra I look forward to all those changes. pwm_mapping.c has caused many headaches. :)

I'll be changing the Sbus inverter to something UI selectable, probably a feature, in the very near future. I hate the idea of giving up a port to sbus only or having to recompile to change that option. I've also thought of making some serial/spi/i2c ports changable without having to recompile. The Revo, for example, shares several features on some of the pins. ADC, I2C, SPI, CAN, Serial and Motor output can all happen on the servo pins.

hydra, 2016-01-18:
@ledvinap See above :smile:

ledvinap, 2016-01-18:
The IO code infers all exti configuration: https://github.com/ledvinap/cleanflight/blob/local/src/main/sensors/initialisation.c#L89-L117

blckmn, 2016-01-18:
@hydra @ledvinap the idea of having a map of IO and its allowable function is nice (UI selectable as per @rs2k). The target would possibly still need to be able to specify what would be permissible for any given pin (or set of pins) to prevent the inadvertent bad config. e.g. if there is a acc/gyro on SPI1 (pin set X) - its highly unlikely it should be configured for anything else.

Should the config be geared around function e.g. i2cX, spiX, pwmX, uartX etc then having appropriate structures for each (defining the allowable pins, clocks, sources etc) by target.

Just thinking out loud :)

ledvinap, 2016-01-18:
@blckmn : With external configuration (cleanflight configurator, etc) all that is necessary is to check for conflicts - that is much easier then trying to resolve them.
Just start initializing everything (without order enforced by conflict resolution), marking used resources. Aborts initialization is resource is already taken. On abort replace configuration with maximum-connectivity default (so that it is easy to reconnect and fix problem) and refuse arming / stay in error state.

blckmn, 2016-01-18:
Makes sense.where do we start :)

ledvinap, 2016-01-18:
@blckmn : You can review/test https://github.com/cleanflight/cleanflight/pull/1421 ...
After that rest of IO stuff needs to be converted to new interface, it should be quite easy.
Timer channels and ADC next ..

hydra, 2016-01-18:
Great to finally see some traction/progress on this stuff.  @blckmn Please keep us posted and keep https://github.com/cleanflight/cleanflight/pull/1421 with your progress.

blckmn, 2016-02-06:
Closing this one as will keep 1421 updated, and pull request exists for starting it off.

closed 2016-02-06 by blckmn