rs2k/raceflight #74: Quanton FC

ther0x, 2016-01-29:
Hi everyone

Is this release compatible with quanton FC?
I'm tring to find out an alternative to Taulabs

Thanks :)

blckmn, 2016-02-02:
We could most likely create a target for it. Are these units still available? Would you be willing to test - I don't have any details on the board (pins outs devices etc).

ther0x, 2016-02-02:
Thanks!
I have those informations

Quanton Pinout/Wiring

https://docs.google.com/spreadsheets/d/1DNAjixNGpjG3VyTk61cjjHc94rMEi0_3QG-DWvZ9cbU/edit#gid=0

And I have asked for F4 support into another fork of cleanflight

https://github.com/sambas/cleanflight 
I have to make make TARGET=COLIBRI but i'm not able

blckmn, 2016-02-03:
I'll look to adding a target in the next 1-2 days. It will need to be tested by you as I don't have the board.

ther0x, 2016-02-03:
Beautiful! Thanks :)

blckmn, 2016-02-05:
I've added a target in the latest pull request.

ther0x, 2016-02-05:
Yeah thanks! I'll try and let you know :)

ther0x, 2016-02-18:
Hi, any news about it?

blckmn, 2016-02-21:
Hi therox. You can pull the latest code and build the Quanton target. It's not tested as I don't have a unit. If you can test and provide feed back we can make any necessary adjustments. Ta.

ther0x, 2016-02-23:
Sorry blckmn...is there an help or tutorial to make it? sorry, I'm a noob... :/

IdefixRC, 2016-03-27:
Here you go:
HEX: https://drive.google.com/file/d/0B8lA7b3y0usFSElWTmo1R0NNaUk/view?usp=sharing
BIN: https://drive.google.com/file/d/0B8lA7b3y0usFVkw4RzRQRnBwYUk/view?usp=sharing

ther0x, 2016-04-29:
Let me try, very busy...sorry :/

closed 2016-06-19 by proggod