rs2k/raceflight #78: Implement LRS Support on Sparky 2

brat002, 2016-01-31:
It will be great if Raceflight will support RFM22b module that comes with Sparky 2.

blckmn, 2016-02-05:
Would the module be of much value racing?

brat002, 2016-02-05:
This module saves place and AUW. Sparky 2 has one reciever. Why we need additonal one?

chocol4te, 2016-02-19:
I am relatively familiar with the OpenLRSng code, and will be happy to work on this feature, but it might be a while before I can buy a Sparky 2.

brycedjohnson, 2016-02-21:
Might be able to get a lower latency tx link by using spi and the RFM22b instead of sbus. Also would need  the opernlrsng to support a serial protocol such as this for opentx (https://github.com/opentx/opentx/issues/2894)

Linjieqiang, 2016-03-28:
This feature sounds nice~

brat002, 2016-03-28:
I am still waiting this feature to flash Raceflight into my Sparky 2.

proggod, 2016-06-19:
Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod