rs2k/raceflight #87: AeroQuad32 V2 and AutoQuad M4 targets in progress

MJ666, 2016-02-15:
I'm actually working on two new F4 targets. The AeroQuad32 V2 target is for SwiftingSpeed from RaceFlight RCG thread. The AQ M4 I have myself but actually mounted in an copter and i have it not connected to the debugger yet. Here is an picture

http://www.multiwii.com/forum/viewtopic.php?f=12&t=4893&start=50#p62382

The AQ M4 is quite interesting since all sensors are hocked up via SPI and it also has CYRF6936 chip wich is used as an DSMX receiver and an uBlox GPS. It has an very small footprint and is mainly an brushed FC but can also be used for larger copters with brushless motors. The receiver code is still far from functional yet. I guess only 80% ready.

http://flyduino.net/Autoquad-M4-v2

Here is an link to the AeroQuad32 board.

http://www.aeroquadstore.com/AeroQuad_32_Flight_Control_Board_Version_2_p/aq32-001.htm

If someone with the hardware, development skills and debug possibilities is interested here are the links to the two development branches.

https://github.com/MJ666/raceflight/tree/AeroQuad_32
https://github.com/MJ666/raceflight/tree/AQ_M4

SwiftingSpeed, 2016-02-17:
As I said on RC Groups, AQ32 both V1 and V2 are practically identical so if someone have only V1 it should be possible to develop on that board too.
The differences are the compass (which is still code compatible) and the physical placement of the pins on the board, V1 have a rather strange placement of all pins, making it necessary to re-order the pins in your servo connectors for example, V2 fixes this and makes it more user friendly.

I wasn't able to get the hex compiled by MJ666 to work on my V1 board but as I lack a hardware debugger I don't know what the problem might be, I'm waiting for one I ordered from eBay.

AQ32 is supported by [Tau Labs](http://taulabs.org/) and [dRonin](http://dronin.org/) if anyone want to look at some code.

MJ666, 2016-02-17:
Have done some updates. @SwiftingSpeed I have placed an updated hex into the RCG RaceFlight thread.

SwiftingSpeed, 2016-02-19:
I tried to get the mag and baro working but was unable to, I used set set baro_hardware=0 but the baro icon in configurator is still not lit so maybe I missed to enable something else? I can't find it as a feature so I don't know where more to look.

If I also set mag_hardware=0 and send save command the board seem to stop communicating and doesn't set up the VCP port correctly anymore, I have to re-flash the board with full chip erase to make it work at all again but then the settings of course are default again.

In the configurator both pid and receiver tabs are unusable so I can't see and verify my receiver channels, when switching to one of those tabs you get stuck at "waiting for data" and can't switch to another tab, you have to click on disconnect and the connect again, no need to restart board or unplug it though.

I have tried motors vi the motors tab and both normal and oneshot esc:s work, don't have any multishot one to test with but I don't see why it shouldn't work as well.

Even though the receivers tab doesn't work I am able to setup an AUX channel for arming and can arm the board via my transmitter, then I can also see that the motors react on board movement and I can also influence them on stick movement as I have the default cleanflight channel order on my transmitter, throttle spins the motors up etc. I have tried with a PPM receiver and also SBUS on UART3, both working.

I have not been able to test any other UART ports at the moment, see if I have the time soon.

Finally I don't know how complicated VCP are supposed to be under Windows but every time the board reboots (when changing port settings for example) I must unplug the USB cable from my computer and plug it again to be able to connect to the board again. Maybe this isn't a raceflight issue, I have read Windows and VCP doesn't work that well.

MJ666, 2016-02-19:
I have checked the sensor part again but did not found an problem here? Assignments look to be correct. Please can you check your board if this have R37 and R38 installed? They should not be there.

The problem of the PID and Receiver page of the configurator was an bug in the last builds. This is not related to the firmware and should be fixed now.

For me all VCP targets will disconnect during reboot and I need to reconnect manually. This is known behavior. If you need to physically disconnect this is not usual be happen with some installation. You should wait a few seconds otherwise it may fail in my experience.

SwiftingSpeed, 2016-02-20:
Resistors R37 and R38 are not populated.

I was now able to connect with mag enabled if I let the board start before connecting the USB cable, either by connecting the USB half-way and letting it startup and then plugging in all the way or by using another power source as a battery/BEC.

I got an UBLOX GPS working on UART2 as well today so the only thing not working now is the baro and as I said, I have already set baro_hardware=0 and now we can also verify i2c is working because of mag working so maybe there's something wrong with my baro sensor?

MJ666, 2016-02-20:
Just rebased the code and made some more updates. The SDCard may now also work? If not please let me know what the CF configurator is reporting at Blackbox page for the SDCard status without an card inserted. May be the card detect pin need to be not inverted. Please also check the sensor orientations carefully before trying to fly. Also verify the motor order with each motor individually in the configurator (if not done already). You also can check the battery monitoring. If everything is going well I will do an PR to merge the code. Minor things can be fixed later on. If the I2C bus is working it is strange if the baro does not work. This really could be an hardware issue? You may try with the original firmware if it works there? I will post an actual hex in the RCG RaceFlight thread.

SwiftingSpeed, 2016-02-21:
I can't get SD card to work, is says no card regardless of card inserted or not.

I managed to load the original firmware and the baro is working there, can clearly see the graph in that configurator changing when blowing onto the sensor so it's not random noise or something like that.

Beeper also seem to not be working. The port is always off while the other external led ports are on so some initialization is maybe working. It makes a short beep when power is connected but I guess that's before the port is configured to be off.

Now the working stuff:
Motor order is verified correct with octocopter (so I could test all 8 outputs).
Battery monitor (ADC1) is working, measures my two test batteries (2S and 3S) correctly with DIY voltage divider.
Sensor orientation seem to be correct from what I can tell, the 3D model in configurator is moving just like I move and lean the board and the compass is showing north when I point the board to north.

It's not exactly flight weather here in Sweden so a real flight is out of the question at the moment and also I don't have a machine ready for the AQ board at the moment as the board was just laying in a box before I realized it maybe was compatible with Raceflight. I will try to setup an indoor test.

MJ666, 2016-02-21:
Does it really says the some thing if card is inserted or not. ok let me disable the card detect pin for now may be this is not working. I would expect something like card is not initialized and card not detected. Now the card should work independent from detection pin.

I might have found a fix fro the baro problem. The I2C address was wrong. Please retest.

For the beeper I was checking the schematics again and I looks to be the LED1 and LED2 ports are reversed?

The board has an internal voltage divider I have reassigned the voltage pin to this (BM+, BM-).

Will post an new build on RCG.

SwiftingSpeed, 2016-02-21:
I checked again as I was unsure I restarted the board in between but yes, it says no card inserted regardless of actual card status. With the new build blackbox is working fine and with no card inserted you get a fatal error in cleanflight configurator but that's expected since you removed the check.

Baro is also working now and the voltage monitor as well with the on board divider.

I don't believe the external led pins are inverted, the processor pins controls a NPN transistor that switches the connected led/buzzer to ground, the other side of the led/buzzer needs to be connected to a power source, 5V for example. The original wiki states that yo should set the pin high when the led is to be lit: http://aeroquad.com/showwiki.php?title=AeroQuad32-Connections#Connecting-LEDs-to-AeroQuad32-v1

If the buzzer pin was inverted, wouldn't the buzzer sound all the time and then be silent for short periods during arming etc? the buzzer is now silent all the time when connected to LED2 port but not when connected to the other led ports. If you are able maybe you can test to switch the buzzer to LED3 port (PE1 instead of PE0) to see that it's not a hardware fault.

I will see if I can build a test quad to make a small indoor test, first by holding it in my hand and then maybe some hovering. It looks really promising and I must thank you again, great work.

MJ666, 2016-02-22:
OK looks to be all real problems are solved and there is may be only some fine-tuning required. I will to an PR for the code now.

I now reactivated the SDCard detect pin. It should work now the inversion likely need to be of.

for the led and the buzzer I don't use inversion. Only for the internal LED's the inversion is used. For the V2 the external LED's 1 and 2 are reversed. So actually the buzzer should be on LED1. I can make an test build for PE1 and post on RCG.

Toogo, 2016-03-21:
I have two AQ M4 's so I looking forward to a test version of the firmware.

Great job!

//Frank

MJ666, 2016-03-21:
@Toogo unfortunately the M4 is still not booting with the code. I even don't get any LED activity and the board is still visible in DFU mode after flashing but not reacting anymore. At the moment I have no idea what is wrong. Also the development of the SPI based sensor drivers is not as successful as I wish. I can communicate with the sensors but the readings are not correct. I think this effort will still take some more time. My M4 is sitting in an small brushed quadcopter and I don't have the debug port connected. Disassembling is not an option here. Getting an other one for hardware debugging would be quite expensive. I also have not found the proper connectors yet for this tiny headers. I fear I only can get forward with hardware debugging at this point. If someone is here and can provide an M4 with the debug connector and the proper cable one an temporary bases it would be helpful to bring the support froward. For supporting the CYRF6936 receiver it will take some more time anyhow.

closed 2016-06-19 by proggod