rs2k/raceflight #98: Incorporate Raceflight into Betaflight

opiswahn, 2016-02-22:
***Disclaimer***
Don't get me wrong, I don't want to start some kind of war here, this is just what i belive would be best for the community.

It would be nice to habe the raceflight changes available in Betaflight. As far as i know, raceflight's only purpose is to have F4 support.
The main problem with that is, that raceflight lags the feature availability from betaflight or even cleanflight. The codebase is in part way older then betaflight or cleanflight, and the porting is very slow if ever possible. This is also due to the issue that raceflight isn't a true fork of betaflight (in GIT terms), so up- or downporting of features is unnessessary complicated.

I really appetiate rs2k for it's work to supply F4's very much, but i think it would serve the community better to have one version supporting all targets with a united dev-power.
That means better testing, better code quality, more devs working for a united goal.

blckmn, 2016-02-22:
There's a lot more in the changes in raceflight than merely F4 support. We are unlikely to get other changes back into clean flight for sometime.

What are your thoughts @rs2k?

opiswahn, 2016-02-22:
Well, that's exactly my point, both race- and betaflight are targeted at creating a low-latency, high-performance version of cleanflight, main difference is that betaflight targets F1/F3, and raceflight targets mainly F4(F1/F3 also, for it was already in the codebase).

Why not combine the efforts, so that all sides profit on that?

borisbstyle, 2016-02-22:
@opiswahn 
Well it is always being combined. @rs2k usually picks some stuff occassionally once its proven to work and other way around. It is completaly normal to have long delays between merging those features.

Supporting many target support is quite hard in terms of support and especially when there are also multiple architectures! What works best for F1 boards may not be best thing for F4 etc...

tracernz, 2016-02-23:
In git terms, git has no concept of forking, if the two branches share common ancestors they can merge. The fact that the branches live in different Github repos is of no consequence except to Github.

blckmn, 2016-02-23:
@tracernz is right. My repo for Raceflight has @borisbstyle , @rs2k and clean flight as remotes. I can pull from either or. Merging on the other hand can be a bit tricky... Or simply cherry pick.

opiswahn, 2016-02-23:
That may be, but the longer those different repos exist, the more the code will diverge. It'll get harder and harder to share or merge code between the projects. 
Even right now code from clean or beta is rarely merged, the other way around even less (everything as far as i know, please correct me if i am wrong). With more progress in each project, code becomes harder to merge due to changes in dependent parts or architectual changes and such.

If lets say beta and race would be 'merged', one indirection would fall(clean<->beta<->race), and the new project would be more up to date with all performance optimization, which are right now done twice or at least separate from each other, and only got merged occasionally (eg. AirMode is still not with all fixes from beta integrated in race)

I also think @rs2k would rater compile new features or tweak performance that merge changes forth and back, only to stay up-to-date with beta or clean :)

blckmn, 2016-02-23:
But that's the thing @opiswahn ... The reason for the fork is current different directions. Raceflight is supporting F4 targets, multi shot etc. these are significant divergent differences that won't in the short term be incorporated in clean flight. I'd say you will see over time Raceflight focussing on quads, racing, and as such the code for other purposes stripped out. You may see F1 support dropped altogether. There's no harm in this happening. Everyone learns ultimately, and the cause is richer for it.

rs2k, 2016-02-26:
RaceFlight's main purpose is to find ways to increase performance on quads. We've been finding the F4s fly completely differently than the F1s and F3s. Filters and code that work well for the F4s never seen to work well for the F1/F3s and vice-versa. This might make reincorporation a bit more difficult, but I still think it would be a good thing. :)

We've reached the practical limit of F1s and possibly even F3s while the F4s still have a lot more potential. For this reason F1 and F3 support has been a bit neglected. It's possible F1 and F3 support will be dropped at some point if it starts to interfere with performance of the better FC targets.

opiswahn, 2016-03-01:
Well, if that's the way to go then the right step would be to dop everything legacy, leave only F4's und future targets(F7 anyone ;)). Concentrate on the important NG stuff, leave the rest to other projects.
F1/F3's are perfectly covered by Betaflight and is very activly developed, no need to do the same stuff here.

hydra, 2016-03-01:
@blckmn  'There's a lot more in the changes in raceflight than merely F4 support. These include driver updates, resource mapping, target file updates, and multi shot. We are unlikely to get these other changes back into clean flight for sometime, possibly never.'

!!!!

I'm mighty confused here - As I've noted before, the only thing stopping cleanflight from having the F4 support has been the resource allocation!  I was just here looking at how that is panning out.

Alas I have had to work on getting other things merged into cleanflight, like scheduler and like a billion other PR's from people, so it has taken a back seat.  If you check the cleanflight issue tracker for the 1.13 milestone you'll see resource allocation is schedule.  That, and the configuration group to further aid decoupling in the codebase are my top priorities.

Likely i will be looking at the raceflight code with regards to the resource allocation and perhaps asking you and @ledvinap the odd question regarding it soon.

I totally agree that merging this into betaflight is a bad idea.  betaflight to me at least, seems like bleeding edge F1/F3 codebase that often has bugs or other problems, vs the focus on raceflight to support F4.

Cleanflight, on the otherhand still stands on it's principles where to try and make it do what users want, but in a maintainable way - and this takes time.

sukoshi1507, 2016-03-01:
Which is the reason why, after testing betaflight on the Gemini, i switched
back to Cleanflight which has been proven to be reliable even without the
geeky bleeding edge part ;-)

proggod, 2016-03-01:
Having so many forks makes it hard for me as a developer to decide which to even work on.  I have F1, F3, F4 boards etc, so any would work for me.  But would be nice to have features reflected through all the different forks.

hydra, 2016-03-01:
@proggod yes, agreed.  i'd prefer making other devs part of the cleanflight team and having them work on branches.  it'd be much easier to keep track of things that way.

hydra, 2016-03-01:
@rs2k and @blckmn this ticket should probably be closed.

nathantsoi, 2016-03-01:
@borisb currently has the best flying and fastest moving codebase :) you
should start there.

@hydra, it would be rad if everyone had push rights to the same repo, but
the master branch should be owned by everyone, not just one dev. e.g. w/
code review and tests, any dev should be able to push to master :)

proggod, 2016-03-01:
I think having one person in charge of master makes sense, so that the person can do code review, and testing before merging into a production branch.    I also don't understand why these changes are completely separate projects and not just different branches.   That would prevent code from getting too out of sync, so merging in features would be simplistic.   If a bug is fixed in one branch, your going to have to manually merge in the fixes into another branch with this current system.

One thing that really attracted me to cleanflight is how organized, and well done it seems to be.

hydra, 2016-03-01:
@nathantsoi that's how it works.  many of the contributors have master branch access and they are all told to PR and review.  I do the PR approach a lot of the time now too, to get other's input on new features before merging.

nathantsoi, 2016-03-01:
i totally agree @proggod :) https://github.com/borisbstyle/betaflight/pull/1

@hydra, that's cool if that's how you want to run things. plus, i think
it's awesome you're willing to do this for everyone, i know it's a ton of
work keeping the code nice, clean and reliable while merging in new
features. im sure it takes a ton of time to go through all those pull
requests :) im just saying to get everyone onboard, i think everyone would
need to have equal rights to merge good code. requiring all the code to be
reviewed / merged by one person is a single point of failure and kinda
defeats the purpose of a shared repo

blckmn, 2016-03-01:
Hey @hydra I'd love to get these changes back into Cleanflight at some point - and understand this takes time - one only has to look at the frequency of pull requests for cleanflight and you can see the difficulty you have on one hand, but the goodness of the sheer number of contributors on the other.

My comment above was no reflection on you or cleanflight - apologies if it came across that way. It was more that @rs2k had indicated his intentions for raceflight where to focus on F4 (likely dropping F1 and possibly also F3 support), but bring in the bleeding edge of betaflight - thus the comment that some of the code here will probably never make it back to cleanflight.

So its really a triangle of love at the moment:
cleanflight, stable support for F3 and F1
betaflight, unstable but bleeding edge changes for F3 and F1
raceflight, sometimes unstable but almost bleeding edge changes for F4.

At the moment I have a love for F4. Raceflight is the only repo I can contribute to that allows me to fly the F4. I have been furiously working away at the timer code to try a few ideas - (some slightly different to @ledvinap) although @ledvinap has done some serious work here - there is still the whole configuration piece missing. Hopefully some of this will make it back to cleanflight. I am certainly happy to contribute time to help get it there too.

tracernz, 2016-03-01:
Having one person effectively a dictator puts me off a project. Need a team in charge of code review, merging and decision making.

borisbstyle, 2016-03-01:
@nathantsoi 
Cheers :dancer:

Well betaflight was actually meant to create ability to do some better public testing and some quick code changing, which you normally don't do. Main focus is flight enhancements and research. The enourmous amount of PR's not getting enough attention in cleanflight was the also main enabler.
To be honest I never expected it to become this big. I try different features and approaches. Some of them fail and some succeed. Eventually I always add those to cleanflight once finished.

But yes true is there are indeed many projects around.

opiswahn, 2016-03-01:
So, how are the plans to handle that? I don't belive that will be a good constallation for future development...
Of course there sould be branches for feature development, improvements or even such big things as F4 support, but i don't really think that should grow into an own project if there isn't a crucial part distinguishing it from it's base.
Developers tend to stay true to their chosen project, so there is litte code transition forth and back between different projects if it is too difficult to merge because of code divergence over time.

I'd think a main project and an experimenting project should be enough, makes code sharing easier and will produce more output if devs can concentrate on one instead of several projects and share their fixes and improvements.

Raceflight is not that active of a project, and i fear that it will get left behing from improvements from clean and beta, simply because there is not enough menpower to transfer and incorperate them. 

Maybe, just my oppinion, it would be better to use that manpower to make clean or beta capable of F4 and grow them to be the best flight controller there is. That can be made on a branch of either of those, but not as a (more or less) independent project.

Again, i don't want to talk down anyones hard work and effort here, without @rs2k @blckmn and the other guys work there may be no F4 support yet, but i think it would be of better use for the community if it got back fast into clean or beta.

hydra, 2016-03-01:
i can add you guys, @rs2k and @blckmn as cleanflight developers, we can create a branch and get all the IO stuff merged in ASAP.  That is number one priority for F4 support since it opens the door for greater flexibility and gives and users and maintainers the flexbility they they need.

proggod, 2016-03-01:
That would be great to see!

nathantsoi, 2016-03-01:
to @tracernz point, how does that fix the problem of "having one person effectively a dictator"? unless you're changing how you run the cleanflight repo, hydra?

proggod, 2016-03-01:
@nathantsoi He said further up there are several contributors with master access.

MJ666, 2016-03-05:
I also would be able to help to get the IO clean up done in cleanflight an get the F4 support. It is fine to have some test platforms to try new things but the final goal in my view shoul be to improve one software. This was my interest from the very beginning when betaflight was just started. We shuld not split up our dev resources to much. Having an small trusted team to make merges will be fine. There is no way everyone can merge. Finaly we want to get a flyable firmware an users rely on this. I'm working with F3 and F4 FC's and want finaly to have an single firmware to use and maintain.

hydra, 2016-03-06:
Agreed 100%

proggod, 2016-03-06:
I'll offer a $200 bounty for someone to merge in f4 support :)

blckmn, 2016-03-06:
@hydra we should probably consider restructuring the project files a bit - improving hardware abstraction. It is possible that arm based MCUs other than STMs will make it into FCs - there's so much conditional code based on defines - specific to targets, let alone MCUs - this is in part what makes merging the most difficult. I'm a culprit for combining drivers in this manner - so I'm going to revert this behaviour, and try and keep MCU / Target specifics out of the core code paths.

You'll see some pull requests from me in cleanflight to help this shortly.

blckmn, 2016-03-06:
Oh and @proggod I might just take you up on that offer! ;)

proggod, 2016-03-06:
@blckmn that would be great! :)

I'd even help, if you want to give me a quick rundown the code..

hydra, 2016-03-06:
@blckmn yea. Gotta keep it clean. Yes. Removing as much of the target specific conditional code as possible would be great.

proggod, 2016-03-06:
So, was spending tonight going through the code, looking at making changes, and I am understanding why we are having these branches all out of sync.  In order to submit code you have to create a fork, which basically takes a snapshot at that moment, and do changes there?   So if it takes you a while to do everything, you end up having a branch well out of sync with everything else?    Is there some reason why we can't branch?

MJ666, 2016-03-06:
You can merge the changes to your master branch and rebase the code in your development branch of your fork. This will allow you to stay in sync with the original master as much as posible and also enable you to create PR against the original master with your changes. What is still an manual process is the resolution of conflicts during the rebase process. This get more and more tricky if the two branches getting more different.

proggod, 2016-03-06:
@MJ666 I did realize last night, that there is still a way to resync with another fork, so thats why i removed my comment.  We use gitlab very heavily internally, so I am use to everyone working in branches, it lets us assign permissions, and even work with each other on the same branch if needed.   I was just not use to the way this is set up, however I think the separate forks leads to a certain mentality, which ends up keep the code from being sync'd into the main fork on a regular basis.  This is probably how we ended up with betaflight and raceflight.

opiswahn, 2016-03-06:
@blckmn if you need some assistance on that task let me know, i'll see what i can do.

rs2k, 2016-03-07:
The reason why we ended up with raceflight is I wanted a way to test my own ideas without stepping on toes. I didn't think @hydra would appreciate me adding thousands of lines of code to a cleanflight branch for a science experiment. :)

The newest raceflight is proving to fly extremely well. The F4s and better are most certainly the right path to go. 8KHz loop times with16KHz PWM rates out of sync is flying considerably better than 4KHz wth oneshot on the F3s. We're talking ultra locked in ETH nearly zero bobble on hard, in propwash manuvers. Wth some more optimization we should be able to do 16 or 32KHz PWMing at 32KHz. I really look forward to seeing if that flies better than 8 to 16.

proggod, 2016-03-08:
Any news on the integration?

MJ666, 2016-03-09:
Fixed all AlienFlight F3 and F4 specific issues cause by the BetaFlight merge. Bench tests are completed without issues. Flight tests are still outstanding.
https://github.com/rs2k/raceflight/pull/110

blckmn, 2016-03-17:
Closing this one.

closed 2016-03-17 by blckmn

proggod, 2016-03-18:
So does that mean its merged in, or have you gave up on the task? :)

blckmn, 2016-03-18:
Sure is. We will try and keep it as up to date as possible. There are still some niggly issues we are working through on the latest merge.

proggod, 2016-03-18:
So is the current version on git compatible with my revo for betaflight? :)   And where do I send the paypal for the $200 :p

blckmn, 2016-03-18:
yup - it should work ok. If you send the coin to my email ... and I'll send you a bluejayf4 in return ;)

proggod, 2016-03-18:
No idea how to get your email, but i messaged you on rcgroups..

rs2k, 2016-03-18:
Great job! I'm compiling this and putting the binaries into the RF dropbox now.

proggod, 2016-03-18:
How do you build, I don't see any targets for F4:

ALL_TARGETS := naze
ALL_TARGETS += cc3d
ALL_TARGETS += cc3d_opbl
ALL_TARGETS += spracingf3
ALL_TARGETS += spracingf3mini
ALL_TARGETS += sparky
ALL_TARGETS += alienflightf1
ALL_TARGETS += alienflightf3
ALL_TARGETS += colibri_race
ALL_TARGETS += lux_race
ALL_TARGETS += motolab
ALL_TARGETS += rmdo
ALL_TARGETS += ircfusionf3
ALL_TARGETS += afromini

I pulled most recent betaflight from github

rs2k, 2016-03-18:
https://www.dropbox.com/sh/dq7z14zpno3zlpe/AAAhHlBnjnVk79RUxvs3REEia?dl=0

blckmn, 2016-03-18:
Merged current betaflight into raceflight, and we will work to keeping it up to date. So if you pull raceflight, it incorporates all the recent betaflight changes.

opiswahn, 2016-03-18:
I think @proggod meant it really other way around :)

proggod, 2016-03-18:
Hah, I guess it doesn't matter I suppose, so is all of beta flight merged in, is it going to be easier to keep everything in sync?

blckmn, 2016-03-18:
Yes - definitely easier. We have to keep on top of it of course. I'll be submitting some pull requests to clean and beta to try and make it a little easier too :)

proggod, 2016-03-18:
Alright great, did you get my message..

blckmn, 2016-03-18:
As in there are consistently filers that'll have merge issues. Breaking them out makes it cleaner to merge.

opiswahn, 2016-03-18:
So, as there will be no backmerge atm, how about dropping at least f1 target's for cleaner code? Those will be fine with beta I think, let's concentate on the performant controllers here.
Might be easier to tweak and tune with no regard to 'legacy' controllers

proggod, 2016-03-18:
I don't know, would be kind of nice to support everything, naze32 and cc3d are still the two most popular boards, and it would be nice to have one software that works on all.

hydra, 2016-03-18:
agreed, it's too early to drop F1 yet.

hydra, 2016-03-24:
to follow up on this.  i've granted @rs2k @blckmn @borisbstyle @ledvinap commit access to the cleanflight firmware repo.  you guys can now create branches, fix bugs, etc.  use PR's to manage changes so we can discuss/schedule/merge as appropriate.

@rs2k is was great catching up with you over in Dubai!  Real shame that @borisbstyle wasn't there.  I'd love to get us all together for some beers at some point.

digitalentity, 2016-03-24:
@hydra great! Although most of my current development happen in INAV repo, I'll slowly start moving some code to a separate branch of CF.

[3 more replies]