rs2k/raceflight #104: SPRACING F3+OSD

gamani, 2016-03-03:
HI Guys,

I've tried to update the firmware of this board SP-Racing-F3-Flight-Controller-With-Integrated-OSD-for-FPV-Multicopter (please google it you'll find the infos)

But it doesn't work.

I think it's pretty easy to support raceflight for it.

Are you able to do this ?

Regards,
Alex

gamani, 2016-03-26:
No answer till now ? great ...

blckmn, 2016-03-26:
Sorry mate. We've just been flat out getting the latest changes from BF working on the F4s.

If this board is a direct clone of SPRF3 then that target should work. If it's not then it might need its own.

It'd be great if you could link all the details here.

gamani, 2016-03-26:
Thanks for answering. In fact its a type of spracing F3 board that we've
developped, if you can give me your e-mail i would be happy forward you all
the details for sure.

blckmn, 2016-03-26:
How many boards are being made? Be good if you could send a sample of one to one of the programmers here to test.

gamani, 2016-03-26:
We're making 100 boards as first batch. But i can send you one of course.

Just give me your address and i will ship it.

gamani, 2016-03-27:
Anyway here are some details 

MPU 6500
BARO BMP 280

The rest is the same as the SPRACING

MJ666, 2016-03-27:
With the last changes on the MPU drivers you can include MPU6050 and MPU6050 driver into the same target.

gamani, 2016-03-27:
Hi @MJ666 ! Yes i've updated the target.h and the makefile but i'm not sure about my changes. Maybe you can have a check ?

TARGET FROM SPRACINGF3
[target.txt](https://github.com/rs2k/raceflight/files/190833/target.txt)

Makefile at SPRACINGF3
[Makefile.txt](https://github.com/rs2k/raceflight/files/190832/Makefile.txt)

MJ666, 2016-03-27:
If you are using the MPU6500 in I2C mode this schould be fine as it is now. What results do you get with this? I'm actualy not at home an can not give this an try to compile. For the MPU9250 there is an additionlal driver update required. likely an  It also should be posible to keep the MPU6050 driver in the target otherwise it will not work anymore with the orginal SPR F3.

gamani, 2016-03-27:
So for 6500 without compass i've managed to compile it, and will test it tomorrow. Then when i add the line "drivers/compass_ak8963.c \" i got the following : ./src/main/drivers/compass_ak8963.c:140: undefined reference to `verifympu6500WriteRegister'. This is strange because on the previous version i've managed to compile it without error.

gamani, 2016-03-27:
Just tested the compiled fw for 6500, well it's initialising but the esc bip are not so sync. The motors don't compensate in airmode and sometimes the board is reinitializing too. Since 2.4.0 update of betaflight i don't manage to make it work anymore. I have a working version on 2.3.5 if you want.

gamani, 2016-03-27:
I think i will leave it to you guys. Any way to send you an FC ? You can send me your full address to atilamusic@gmail.com Also if you have skype or whatsapp, would be great too :)

blckmn, 2016-03-27:
Just sent you an email.

gamani, 2016-03-27:
Ok got it, just replied you ;)

MJ666, 2016-03-27:
Betaflight is bit different to Raceflight. The AK8963 driver needs some tweaking. Actualy it expects if SPI is enabled it is part of an SPI connected MPU9250. It looks to be you already get attention from @blckmn.

blckmn, 2016-03-27:
@MJ666 we might need to add in some specifics for how the AK8963 is connected perhaps a check for a 9250 first...

In this case I'm assuming the AK is connected via I2C...

Or we go with a specific 9250 driver... Thoughts?

MJ666, 2016-03-27:
I think we likely need an specific MPU9250 I2C driver. This driver need to enable the I2C bypass. Cleanfligth has an working implementation for this for the SP Racing F3 mini. The AK8963 driver also need some tweaking. The SPI section should only be compiled in if the SPI MPU9250 is also present.

MJ666, 2016-03-29:
@gamani please can you give this branch an try: https://github.com/MJ666/raceflight/tree/Zigogne

MJ666, 2016-04-04:
@gamani  @blckmn My approach here was not to add an additional target but update the SPRACING3 target to support this controller since this is quite similar. 

https://github.com/rs2k/raceflight/pull/130

blckmn, 2016-04-04:
Hi @MJ666 I did see that ... though I added an extra target to give the target board identifiers (so the imagery can be right). There's also possibly a serial pass through required for the OSD. 

I think we will eventually move away from somewhat away from specific targets - and focus on the chips installed (and how they are connected). Everything else will be configuration downloaded to the board.

blckmn, 2016-04-04:
We can consolidate. Did you get the dodo working?

MJ666, 2016-04-04:
I bench tested this on one of my dodo boards and it looks quite well. I have not done a flight test.

nathantsoi, 2016-04-04:
maybe this is a lame question, how is this board different from the others currently available? e.g. 
- http://nathan.vertile.com/blog/2016/02/03/rctimer-beerotorf3-flight-controller/
- http://shop.myairbot.com/index.php/flip32f3aio-lite.html
- http://www.banggood.com/SP-Racing-F3-Flight-Controller-With-Integrated-OSD-for-FPV-Multicopter-p-1032369.html

henn1001, 2016-04-04:
I think the topic here is about the third FC you linked..

W3DRK, 2016-04-04:
I was just pointed to this thread a few minutes ago...  I actually tried rolling my own build of Betaflight 2.6.0 for this FC last night but didn't have much luck.  I did basically the same as suggested here, I duplicated the SPRF3 to a new target and swapped the gyro/accel parameters out for the MCU6500.  It compiles fine and when flashed the FC seems to work for about 10 seconds, then freezes.  If I keep it moving, it will continue to work, but if I hold it still after a few seconds it stops responding to the VCP. I didn't get a chance to test the motor controls.

blckmn, 2016-04-04:
This one doesn't use VCP.

The drivers in Raceflight have had some work on them too ... So there are some underlying code differences. We have been quite active in that space.

proggod, 2016-06-19:
Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod