rs2k/raceflight #108: REVO issues

opiswahn, 2016-03-08:
This is just a summary of what's working on my REVO(china clone i think) and what not. 
If anything works or works not for you differently, write a quick comment.
I will update this issue to new changes.

Working:
- ACC
- Gyro
- Baro

NOT working:
- GPS
- Led
- Mag

rs2k, 2016-03-08:
Thanks! When I get back from Dubai I'll plan on looking into this.

kc10kevin, 2016-03-10:
Did you try the LED strip on S6_Out on the Revo instead of S5_Out?  From what I saw in the LED strip code, its looking for it on PA0 (S6_Out on the revo).   Am I reading the code wrong?

opiswahn, 2016-03-10:
It's on PWM6(PA0),which is Pin6. Dosn't work, dunno why. Doesn't work on NAZE either... 
I thought it das something to do with the DMA settings on F4 at first, but as it doesn't work on NAZE either, it has to be a more basic problem.

kc10kevin, 2016-03-10:
Got me then.   I just thought I'd throw that out since it might get confusing.   I'll have to get my Revo out of the box and take a look at it this weekend.

cooked, 2016-03-28:
Hey, on my REVO just ACC and GYRO are working... no BARO, no MAG

opiswahn, 2016-03-28:
Baro and Mag are just disabled by default. You can enable them in CLI:
set baro_hardware=0
set mag_hardware=0

cooked, 2016-03-28:
thanks @opiswahn , noob here :)... can confirm ACC, GYRO, BARO, MAG working on my revo. however I experience unexpected disconnection of the board each time I navigate away from the CLI page in the configurator (even if the CLI has not been used)...should I open another issue?
cheers,
Ste

MJ666, 2016-03-28:
For all targets wich use VCP (build in USB Virtual Com Port) the configurator disconnects during the reboot of the FC. This is the case since the VCP will disappear during the reboot. This is actually normal behavior an you manually need to reconnect.

cooked, 2016-03-28:
I don't see this as a normal behaviour.. why the configurator would reboot the board just because I navigate away from the CLI?

MJ666, 2016-03-28:
I can understand you don't like this behavior but the FC need to be rebooted to leave the CLI mode and get back into the MSP mode to be able to communicate again with the configurator. The configurator could also automatically reconnect after a while in this situation. At the moment there are many more important things to improve and fix in the firmware.

proggod, 2016-06-19:
Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod