rs2k/raceflight #109: SPARKY2 issues

opiswahn, 2016-03-08:
This is just a summary of what's working on my Sparky2 and what not. 
If anything works or works not for you differently, write a quick comment.
I will update this issue to new changes.

Working:
- ACC
- Gyro

NOT working:
- GPS
- Led
- Mag
- Baro

rs2k, 2016-03-08:
Thanks! When I get back from Dubai I'll plan on looking into this.

The Sparky 2 has a few speed / CLI / MSP issues that need to be resolved as well.

blckmn, 2016-03-17:
I am finding that the latest betaflight merge the serial is slow for the first 10-15 seconds after a reboot, after that it hums - so there is something in it.

MJ666, 2016-03-18:
We find with an build based on the latest master it takes about 15s before the FC is ready to get armed. During this time one LED is blinking in about an 1s rhythm before it goes of.

blckmn, 2016-03-24:
@MJ666 the delay is something to do with the scheduler. I am trying to find it :)

rs2k, 2016-03-24:
Ugh, that task scheduler is always slowin' us down. lol

StraTech, 2016-05-08:
I was a able to enable the baro through cli by changing (sorry I forgot the prefix)  ???baro=1 to set ???baro=0.
On another subject I was wondering if Raceflight on a SPARKY2 supports PPM or IBUS from a IA6B reciever?

proggod, 2016-06-19:
Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod