rs2k/raceflight #112: List of 'main targets'

opiswahn, 2016-03-11:
We should make a list of 'main targets', targets which are mostly used by raceflight users and more importantly on which new versions can be tested on hardware. For those targets then it can be said that all new versions are hardware tested when new versions are released.
This should make it easier to provide stabler release, since the main targets could be tested on the specific hardware.
All other targets are then only supported as 'compile tested'

I would propose all F4 targets as main target.

As for available hardware, i have: (will complete the list if people are providing themselfes as tester :;) )
- Revo
- Sparky2
- Naze(flip32)

mikecarr, 2016-03-11:
Are we no longer targeting Revo Nano? Would Revo clones fall into Revo?

opiswahn, 2016-03-11:
Sure, RevoNano is still a target, but there seem to be several problems with it as discussed in the forum. I don't know the current status of the target,

As of the other point, i would count the revo clones as REVO as long as there are no known issues with the clones (which there aren't as far as i know).

mikecarr, 2016-03-11:
There is one Revo clone that appears as Revo Nano by Far West Rotors
http://www.farwestrotors.com/#farwestrotors, I will talk with rs2k to find
out the plan with that.

rs2k, 2016-03-11:
The Revo Nano has been on the back burner for a while now. I hope to get it working correctly when I get back from Dubai. Once the problems with the Revo Nano are resolved I really don't think STM32F411 support will take away from project.

proggod, 2016-06-19:
Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod