rs2k/raceflight #116: Serial 1wire BLHeli passthrough via VCP

blckmn, 2016-03-17:
@tracernz I know you're good in this area. Happy to help out here to get this working on the F4 targets using VCP.

Thoughts?

tracernz, 2016-03-18:
Ah, the cleanflight code is way too messy to the point where it's a real chore to do this kind of stuff, I don't enjoy it. Raceflight currently has sambas's implementation I think which should work, or be very close.

tracernz, 2016-04-10:
4712 has taken some of the code I started with and written two different implementations which seem to work well. One a simple passthrough like the existing, and one that supports his 4-way protocol for both BLHeli and SimonK bootloaders. If you pick them up you can close this. ;-)

blckmn, 2016-04-10:
https://github.com/rs2k/raceflight/pull/142 just put this together for the 4way (@4712)

4712, 2016-04-10:
@tracernz yeah thanks for implementing the [pwmEnableMotors](https://github.com/4712/cleanflight/commit/47348dd696ab7938043f35e6a21ee2d25fb05a18#diff-f7f635af239845c0d74ed943bd7aff80R53)() realization. It gives some more security, to prevent motors from running, while working with the ESC.
Also thanks a lot for your [usbVcpGetBaudRate(serialPort_t *instance)](https://github.com/4712/cleanflight/commit/47348dd696ab7938043f35e6a21ee2d25fb05a18#diff-d39cd1eaef49ef86f24def4479928f63R186)! It was essential for getting 1-wire-pass-through-vcp channel switching working. Awesome Idea!!! 
Although the 4way-interface does no longer use baudrate feature for the switching, I did not remove it, because I had the idea, that it could be useful for other opportunities.

I realized, that cleanflight integration of SoftSerial, what you tried to use for the VCP implementation, was not adequate for the task. Mainly because it could not switch fast enough from transmit to receive on the ESC side. So, I used a port of my AVR soft serial stuff ([which ](https://github.com/4712/cleanflight/commit/47348dd696ab7938043f35e6a21ee2d25fb05a18#diff-2220e41c0a901046f2d883868079c486R107)is/was equal to the soft serial part of the current 4way-interface). 
May be you or someone else will find a way to speed up the existing cf softserial implementation, so that it can be used also for the 4way-interface.

4712, 2016-04-10:
@blckmn I'm still waiting for a Revolution board, I ordered 2 weeks ago. As soon as I get it, I will test...

tracernz, 2016-04-10:
@4712 I wondered if that was the problem (soft-serial not switching direction fast enough). I will see what I can do for dRonin with a different (new) soft-serial driver.

4712, 2016-04-10:
@tracernz While testing with softserial, I found another issue: the bytes where send out, but nothing came back. 
You need to set the ESC line as output only as long as you're sending bits, mediately afterwards set ESC to input with pull up activated. One more: Shifting out the stop bit first, saves time for switching to receive mode.

tracernz, 2016-04-10:
@4712 Thanks, I will keep those in mind. :+1:

blckmn, 2016-04-19:
This has been merged into RF through https://github.com/rs2k/raceflight/pull/142

closed 2016-04-19 by blckmn