rs2k/raceflight #122: Latest version doesn't work with sparky2

proggod, 2016-03-23:
Latest version doesn't work with sparky2, can't even make motors spin from motors tab.  Latest stable motors work, however can not seem to get DSMX sat to work, althought that might be a lack of an inverter, however it locks the board completely everytime I switch tabs in configurator.   I have to pull usb and plug it back in to make it work.

proggod, 2016-03-23:
Got SBUS atleast working with sparky2, but still the bug where motors do not work at all in current build, also it tends not to lock up on a mac..

MJ666, 2016-03-23:
What protocol you are using fro the ESC's. Oneshot125 with Kiss ESC's was also not working for me with latest master or RC8. ESC's don't initialize. You can try the code from this PR https://github.com/rs2k/raceflight/pull/117 is this is the case for you. There is also an version based on RC8 with the PWM changes https://github.com/MJ666/raceflight/tree/AFF4_RC8 (this version will not work well with PPM receivers).

opiswahn, 2016-03-23:
Seems to be a problem with Oneshot, my sparky2 rig has no oneshot as the ESCS doesn't support it, but it flyes fine, no problems whatsoever with the ESCs. If I put the sparky in my other rig with oneshot ESC's,  it seems the ESCs aren't initializing. Same on naze (flip32), pwm is fine, oneshot not.

proggod, 2016-03-23:
Ya using oneshot, although I think i tried without oneshot and still didn't work.   Just tried it with sparky (non 2.0) and its having similiar very weird issues.  Can't calibrate the esc's at all, and when i try and enable them in motor tab they go right to full throttle.

MJ666, 2016-03-23:
This is what I found with latest master and RC8. Did you try the updated code I reference above?

proggod, 2016-03-23:
No how do I pull that?   My sparky2 just completely stopped working now, won't even power up from usb.  I guess I am going back to the revo.  I could still test it on sparky 1.0 if you want

MJ666, 2016-03-23:
As long the USB is still in position it will be more easy to solder it back.

proggod, 2016-03-23:
I have the build environment, but is there a branch on the main repository i can just switch to and use git to update it? 

No its not receiving power from the BEC either.  Its completely dead, which is completely crazy.

Along the same issues with motors tab, I am running the stable build on the revo now (i replaced it with a revo) and I can not get the ESC to go into calibration mode.   They are little bee 20a.   I set limit to 2000, raised up on motor tab, and plugged in power like I have done a hundred times, and no special tones.

blckmn, 2016-03-23:
The master branch has a corruption issue on the configuration following the merge of the latest Betaflight changes. It affects ESC outputs. Once @MJ666 merges his PWM changes, I'll merge in the fix. We can go from there.

You could pull my PR into your local and try it too if you want a crack earlier.

proggod, 2016-03-23:
To use your PR, do I have to git it from your github?   Stable seems to be working ok with multi-shot, although oneshot is definitely broken, so I can stick with that till the fixes are merged in.

blckmn, 2016-03-23:
You can just create a local branch and pull the PR from the rs2k repo

git fetch origin pull/ID/head:BRANCHNAME

Change ID to the PR id and Branchname to the branch name you want to give it locally

blckmn, 2016-03-23:
Origin being the name you have for the rs2k remote (in my case I called it rs2k for want of a better name lol)

proggod, 2016-03-23:
I know how to use git :)  BUt what I am asking is what branch on what repo?   Is it rs2k on the main raceflight repo?

blckmn, 2016-03-23:
That's the branch it's the branch called 'pull/ID/head' where I'd is the PR id- I think it's 118 - so it's 'pull/118/head'

Note I have not test compiled sparky 2 so there might be some compilation issues needing to be resolved as my PR is a discussion one... With a few fixes.

blckmn, 2016-03-23:
Repo is rs2k/Raceflight

proggod, 2016-03-23:
Alright now I got it thanks, we don't use these PRs internally, so I didn't realize I had to specify it as well.

MJ666, 2016-03-23:
I merged PR #117.

proggod, 2016-03-24:
Well my sparky2 was fried, but tried the latest with a sparky.   It won't calibrate at all, doesn't go into calibration mode, and won't even spin the motors.

MJ666, 2016-03-24:
What settings you are using for onshoot125? What ESC's you are using? Have you tried this?

feature ONESHOT125
feature USE_PWM_RATE
set motor_pwm_rate = 2000

proggod, 2016-03-24:
Yes those are exact settings I tried, and tried with little bees and emax nanos.

proggod, 2016-03-24:
Cleanflight worked fine

blckmn, 2016-03-24:
Ok - just created a PR for some more of betaflight, and the latest PR118 has now been rebased.

@proggod I'd recommend PR 118 and the following:

set motor_pwm_rate=400
set motor_pwm_protocol=125
save

and see whether that works?

proggod, 2016-03-26:
Tried it, made the tones to go in calibration mode immediately with a sparky when I go to use the motors tab.  The minute I plugged in battery it made the tones..

proggod, 2016-06-19:
Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod