rs2k/raceflight #123: 能否在下一版本中加入fs-i6接收机的ibus的支持

pdwdw, 2016-03-23:
能否在下一版本中加入fs-i6接收机的ibus的支持

sukoshi1507, 2016-03-23:
It works as-is if you use PWM.

You mean using the flysky i-bus serial protocol ?

sukoshi1507, 2016-03-23:
(And it should work using serial protocol since it's already supported : https://github.com/rs2k/raceflight/tree/raceflight/src/main/rx )

opiswahn, 2016-03-23:
It works, I am using it myself. But you have to set it in CLI, as it us nut supported in the configuration yet.

opiswahn, 2016-03-23:
set serialrx_provider=ibus

pdwdw, 2016-03-24:
谢谢给位大神的指点，下午我去试试，希望以后能更加完善ibus，fs-i6这个控现在可是无敌了，fs-i6s月底也能到手，真是一代神控了

pdwdw, 2016-03-24:
@opiswahn 下午按照说的在CLI里边对ibus经行了设置，接上接收机后在遥控设置界面里边没有任何反应，换了其它串口也是如此，对了我的飞空硬件是cc3d revo

PM said that in accordance with the inside of the CLI ibus been set, after the receiver connected to the remote setup interface inside no response, for the other serial too, for my fly empty hardware is cc3d revo

pdwdw, 2016-03-24:
[image]
[image]
[image]
[image]
[image]

opiswahn, 2016-03-24:
I am not exactly sure what you mean, the translation is not that good :)
USART1 is on main-port, USART3 is on flexi-port

pdwdw, 2016-03-24:
@opiswahn 所有的usart我都设置好了并一一试过了，但是在界面里遥控并没有任何反应，我保证接线没有错误
谢谢

Jahnkeanater, 2016-04-01:
Whats the easiest way to hook up a serial rx? Is uart 6 in the oplink air port? The oplink port has 7 wires so gnd, vcc, rx, tx... What are the last 3 pins? It would be nice to use uart 6 and keep uart 1 and 3 open for osd and config.

zipray, 2016-04-30:
@pdwdw 你测试过PWM模式吗？PWM各通道都是好的吗？

proggod, 2016-06-19:
Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod