rs2k/raceflight #126: 10-15 second delay before Scheduler Settles

blckmn, 2016-03-25:
Have narrowed this down to GYRO PID task. for 10+seconds on startup it takes 1609 micros to execute. Following that it moves to 124 micros. everything else is delayed whilst this occurs... hence the slow serial.

blckmn, 2016-03-25:
It is the SDCARD afatfs_poll(); that is causing the issue. #undef SDCARD and the settling is immediate.

blckmn, 2016-03-25:
Is there any reason these tasks are in the main gyro pid loop?

I feel they should have their own task scheduled.

MJ666, 2016-03-25:
I hHve done some testing. Disable SDCard support wit actual master will avoid the startup delay but not the strange motor behavior during arming on the brushless. This build is still not flyable. I also retested an build immediately after the SDCard merge I did. This build is behaving normal no startup delay and no motor issues. i can fly this normally. My test build is freshly compiled from this branch: https://github.com/MJ666/raceflight/tree/SDCard_update

blckmn, 2016-03-25:
@MJ666 can you try my latest PWM rework with the brushless - and SDCARD disabled?

MJ666, 2016-03-25:
@blckmn Just updated my earlier comment. Also the build I was refereeing  is not really flying well. 

Will try yours now.

MJ666, 2016-03-25:
@blckmn the AlienFlightF4 looks up as soon I try to configure for oneshot. using the fllowing commands

    set motor_pwm_protocol=125
    set motor_pwm_rate=2000
    save

One led stays steady and the comport is not coming back again. I only can get it back loading an firmware which is overwriting the EEProm configuration in DFU mode via the boot pins. I isolated this issue to the motor_pwm_protocol command.

blckmn, 2016-03-25:
That'll be the lookup table indexes off somewhere. I'll look to see what AFF4 has defined and see what's missing.

MJ666, 2016-03-25:
this does happen after the fc is rebooting. It is still responding properly to the CLI as long I don't save and rebooot.

blckmn, 2016-03-25:
@MJ666 it is booting, initialising etc - but the VCP is failing now. The board is in the main loop, updating motors etc - but the VCP is not running.

blckmn, 2016-03-25:
VCP is back - but chrome locks up on connection.

rs2k, 2016-03-26:
@blckmn it's probably an MSP packet causing it. See if a serial terminal works with CLI.

MJ666, 2016-03-26:
For me the serial port is not comming back after reboot of the FC.

blckmn, 2016-03-26:
I've found the problem, it's a hard fault if 125 is enabled and SDCRD is not defined. Something a miss, but I'll track down the exact cause shortly.

blckmn, 2016-03-26:
Sorted. My bad. Configuring the motors AFTER assigning the write function.

MJ666, 2016-03-26:
This part is now working again but no real change for the flight behavior. Will be out for the next few days and can not do more test till Tuesday.

kc10kevin, 2016-03-31:
Same issue on the KKNGF4 SDCARD version.   So disabling the detect exti code will solve the issue?   Just checking since the branch you mention is over 30 days old.

kc10kevin, 2016-04-03:
FYI.  This issue is present in Betaflight as well.  On the KKNGF3 with SDCard, it is present without the SDCard.  With the SDCard in the slot, it does not happen.  Same behavior in the KKNGF4 w/SDCard.

MJ666, 2016-04-03:
You are using an card detect pin? I this low or high if the card is inserted?

kc10kevin, 2016-04-03:
Yes.  Pin is low when card inserted.

blckmn, 2016-04-10:
This is an issue with card detection. I'll close it - doesn't occur with the current Raceflight

closed 2016-04-10 by blckmn