rs2k/raceflight #127: TBS Colibri support (Gemini)

IdefixRC, 2016-03-25:
Hi guys,

Would it be possible to add support for the TBS Gemini Colibri FC?
Sambas original port of CF for the STM32F4 included the code for it. Unfortunately he has since stopped developing it further. Also my attempts to get the old code updated/uplifted to latest CF have not been successful so far (lots of changes required in the driver code).

Thanks !

IdefixRC, 2016-03-25:
Hi guys, have started here to add the Colibri specific adaptations: https://github.com/IdefixRC/raceflight/tree/Colibri

Following files will need special attention as they included colibri specific code before:
- bus_spi.c
- serial_uart_stmf4.c
- Target folder - Colibri

MJ666, 2016-03-25:
Is the Colobri not an smaler variant of the Quanton board and should be supported by this target?

IdefixRC, 2016-03-26:
Good hint. Let me compare those 2 and see if it can be reused.

IdefixRC, 2016-03-27:
Hi guys,

I checked the quanton and compared the code to SAMBAS original Gemini code. Looks different - e.g. pinout.

I continued to work on the code based on Sambas original and got it to work partially.
Configurator works, calibration works, sensors go into the right direction and Motors 1,2,3 & 5 are working.

Immediate problem areas:
- I have trouble with motors 4 & 6. No movement at all. I checked the PWM mapping and it looks ok.
- PPM RX is not working at all - no movement what so ever

A bit lost where to look. If someone could give me a hint? or take a look at the code (https://github.com/IdefixRC/raceflight/tree/Colibri). Thanks !

blckmn, 2016-03-27:
You might have motor 5 and 4 mapping around the wrong way. 4 motors working seems right if you've set it to QuadX ;).

IdefixRC, 2016-03-28:
Thanks for the reply. Would be great if we can get this to work :-)

The mixer is set to HEX in the GUI (the Gemini is a Hex) so I would expect all 6 to work (or is the HEX mix not implemented in raceflight? I saw the Quanton has a flag for QUAD MIX only in the target.h - the Colibri target.h does not have that). Is the motor mapping done in pwm_mapping ? If yes, I have it follow SAMBAS Gemini Colibri mapping so that should be fine I think.

As for the PPM RX input, which area should I look at to trouble shoot that one?

Thanks a lot !

MJ666, 2016-03-28:
For motor mapping there a two sections responsible. One is in timer.c and the second is in pwm_mapping.c. for your PPM RX check the proper use and assignment of the ppmAvoidPWMTimerClash() function in pwm_mapping.c.

IdefixRC, 2016-03-28:
Thanks. 
I had a look and tried to set it accordingly by adding colibri specific code. No joy I'm afraid. Poking around in the dark here to be honest :-( comparing this with both Sambas CF 1.10 code and the not fully working betaflight conversion, I can not see that I missed something in timer.c or pwm_mapping.c from a pure Colibri point of view.

I might have missed something somewhere else. :-(

nwenkel, 2016-03-29:
I would like to see the support for the Gemini FC too and yes the Colibri FC is a variant of the Quanton board. I have tested the Quanton a while ago but sold it later on. 
Please keep on going.

Kind regards out of Germany

Norbert

IdefixRC, 2016-03-29:
Hi guys,

I have played around more and tried to trace back where the issue might be. With my limited knowledge I'm unfortunately at the end of things. @MJ666 and @blckmn if you find the time, can you have a look at the code below please or my source (https://github.com/IdefixRC/raceflight/tree/Colibri) ? Thanks a million 

Timer.c
I played around with the 2 outputs which I believe are motors 4 and 6.
Commented line is the original, used line is from the Quanton code. Same result in both instances.

`#if defined(COLIBRI)
const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
    { TIM1, GPIOA, Pin_10, TIM_Channel_3, TIM1_CC_IRQn, 0, GPIO_Mode_AF, GPIO_PinSource10, GPIO_AF_TIM1},    			// S1_IN - PA10
    { TIM8, GPIOC, Pin_6, TIM_Channel_1, TIM8_CC_IRQn, 0, GPIO_Mode_AF, GPIO_PinSource6, GPIO_AF_TIM8}, 				// S2_IN - PC6
    { TIM8, GPIOC, Pin_7, TIM_Channel_2, TIM8_CC_IRQn, 0, GPIO_Mode_AF, GPIO_PinSource7, GPIO_AF_TIM8}, 				// S3_IN - PC7
    { TIM8, GPIOC, Pin_8, TIM_Channel_3, TIM8_CC_IRQn, 0, GPIO_Mode_AF, GPIO_PinSource8, GPIO_AF_TIM8}, 				// S4_IN - PC8
    { TIM2, GPIOA, Pin_15, TIM_Channel_1, TIM2_IRQn, 0, GPIO_Mode_AF, GPIO_PinSource15, GPIO_AF_TIM2},   				// S5_IN - PA15
    { TIM2, GPIOB, Pin_3, TIM_Channel_2, TIM2_IRQn, 0, GPIO_Mode_AF, GPIO_PinSource3, GPIO_AF_TIM2},   					// S6_IN - PB3
    { TIM5, GPIOA, Pin_0, TIM_Channel_1, TIM5_IRQn, 0, GPIO_Mode_AF, GPIO_PinSource0, GPIO_AF_TIM5},   					// S7_IN - PA0
    { TIM5, GPIOA, Pin_1, TIM_Channel_2, TIM5_IRQn, 0, GPIO_Mode_AF, GPIO_PinSource1, GPIO_AF_TIM5},   					// S8_IN - PA1

    { TIM3, GPIOB, Pin_0, TIM_Channel_3, TIM3_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource0, GPIO_AF_TIM3},    				// S1_OUT - PB0
    { TIM3, GPIOB, Pin_4, TIM_Channel_1, TIM3_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource4, GPIO_AF_TIM3},    				// S2_OUT - PB4
    { TIM3, GPIOB, Pin_1, TIM_Channel_4, TIM3_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource1, GPIO_AF_TIM3},    				// S3_OUT - PB1
    // { TIM12, GPIOB, Pin_15, TIM_Channel_2, TIM8_BRK_TIM12_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource15, GPIO_AF_TIM12}, 		// S4_OUT - PB15
	{ TIM12, GPIOB, Pin_15, TIM_Channel_2, TIM8_BRK_TIM12_IRQn, 0, GPIO_Mode_AF, GPIO_PinSource15, GPIO_AF_TIM12}, 		// S4_OUT - PB15
    { TIM3, GPIOB, Pin_5, TIM_Channel_2, TIM3_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource5, GPIO_AF_TIM3},    				// S5_OUT - PB5
    // { TIM12, GPIOB, Pin_14, TIM_Channel_1, TIM8_BRK_TIM12_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource14, GPIO_AF_TIM12}, 		// S6_OUT - PB14
	{ TIM12, GPIOB, Pin_14, TIM_Channel_1, TIM8_BRK_TIM12_IRQn, 0, GPIO_Mode_AF, GPIO_PinSource14, GPIO_AF_TIM12}, 		// S6_OUT - PB14
    { TIM10, GPIOB, Pin_8, TIM_Channel_1, TIM1_UP_TIM10_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource8, GPIO_AF_TIM10},    		// S7_OUT - PB8
    { TIM11, GPIOB, Pin_9, TIM_Channel_1, TIM1_TRG_COM_TIM11_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource9, GPIO_AF_TIM11},	// S8_OUT - PB9
};

#define USED_TIMERS  ( TIM_N(1) | TIM_N(2) | TIM_N(3) | TIM_N(5) | TIM_N(12) | TIM_N(8) | TIM_N(10) | TIM_N(11))

#define TIMER_APB1_PERIPHERALS (RCC_APB1Periph_TIM2 | RCC_APB1Periph_TIM3 | RCC_APB1Periph_TIM5 | RCC_APB1Periph_TIM12 | RCC_AHB1Periph_GPIOA | RCC_AHB1Periph_GPIOB | RCC_AHB1Periph_GPIOC)
#define TIMER_APB2_PERIPHERALS (RCC_APB2Periph_TIM1 | RCC_APB2Periph_TIM8 | RCC_APB2Periph_TIM10 | RCC_APB2Periph_TIM11 )

#endif`

pwm_mapping.c
my take is that the first section multippm is the relevant code. I took all the other definitions out which did not match the Quanton code. Same outcome.

`#ifdef COLIBRI

static const uint16_t multiPPM[] = {
    PWM1  | (MAP_TO_PPM_INPUT << 8),     // PPM input
    PWM9  | (MAP_TO_MOTOR_OUTPUT << 8),      // Swap to servo if needed
    PWM10 | (MAP_TO_MOTOR_OUTPUT << 8),      // Swap to servo if needed
    PWM11 | (MAP_TO_MOTOR_OUTPUT << 8),
    PWM12 | (MAP_TO_MOTOR_OUTPUT << 8),
    PWM13 | (MAP_TO_MOTOR_OUTPUT << 8),
    PWM14 | (MAP_TO_MOTOR_OUTPUT << 8),		// USART2
    PWM15 | (MAP_TO_MOTOR_OUTPUT << 8),     // USART2 // Swap to servo if needed
    PWM16 | (MAP_TO_MOTOR_OUTPUT << 8),      // Swap to servo if needed
    // PWM2  | (MAP_TO_MOTOR_OUTPUT << 8),      // Swap to servo if needed
    // PWM3  | (MAP_TO_MOTOR_OUTPUT << 8),      // Swap to servo if needed
    // PWM4  | (MAP_TO_MOTOR_OUTPUT << 8),      // Swap to servo if needed
    // PWM5  | (MAP_TO_SERVO_OUTPUT << 8),      // Swap to servo if needed
    // PWM6  | (MAP_TO_SERVO_OUTPUT << 8),      // Swap to servo if needed
    // PWM7  | (MAP_TO_SERVO_OUTPUT << 8),      // Swap to servo if needed
    // PWM8  | (MAP_TO_SERVO_OUTPUT << 8),      // Swap to servo if needed
    0xFFFF
};

	
static const uint16_t multiPWM[] = {
    PWM1  | (MAP_TO_PWM_INPUT << 8),     // input #1
    PWM2  | (MAP_TO_PWM_INPUT << 8),
    PWM3  | (MAP_TO_PWM_INPUT << 8),
    PWM4  | (MAP_TO_PWM_INPUT << 8),
    PWM5  | (MAP_TO_PWM_INPUT << 8),
[... 148 lines cut ...]

#define SENSORS_SET (SENSOR_ACC|SENSOR_MAG|SENSOR_BARO)

#define USE_ADC
#define VBAT_ADC_PIN                PC0
#define VBAT_ADC_CHANNEL            ADC_Channel_0

#define LED_STRIP
#define LED_STRIP_TIMER TIM4

//#define GPS
#define BLACKBOX
#define TELEMETRY
#define SERIAL_RX
#define AUTOTUNE
// #define USE_QUAD_MIXER_ONLY
#define USE_SERVOS
#define USE_CLI

#define USE_QUATERNION

#define TARGET_IO_PORTA 0xffff
#define TARGET_IO_PORTB 0xffff
#define TARGET_IO_PORTC 0xffff
#define TARGET_IO_PORTD 0xffff`

MJ666, 2016-03-30:
This looks to be the timer mapping Taulabs is using fro the colibri target:

    /*
     * 	OUTPUTS with extra outputs on receiverport
    	1:  TIM3_CH1  (PB4)
    	2:  TIM3_CH2  (PB5)
    	3:  TIM3_CH3  (PB0)
    	4:  TIM3_CH4  (PB1)
    	5:  TIM12_CH1 (PB14)
    	6:  TIM12_CH2 (PB15)
    	7:  TIM10_CH1 (PB8)
    	8:  TIM11_CH1 (PB9)
    	9:  TIM8_CH1  (PC6)		(IN2)
    	10: TIM8_CH2  (PC7)		(IN3)
    	11: TIM8_CH3  (PC8)		(IN4)
    	12: TIM2_CH1  (PA15)	(IN5)
    	13: TIM2_CH2  (PB3)		(IN6)
    	14: TIM5_CH1  (PA0)		(IN7)
    	15: TIM5_CH2  (PA1)		(IN8)
     */

You can find more HW related info here:

https://github.com/TauLabs/TauLabs/blob/20150922/flight/targets/colibri/board-info/board_hw_defs.c

IdefixRC, 2016-04-09:
@MJ666 , thanks for the pointers.
I have checked all and confirmed they are correct. When moving the timers according to the taulabs code I managed to get motors 1-4 to work but not 5 & 6.
Looking at the code I feel the issue is with the 2 TIM12 channels:

`
5: TIM12_CH1 (PB14)
{ TIM12, GPIOB, Pin_14, TIM_Channel_1, TIM8_BRK_TIM12_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource14, GPIO_AF_TIM12}, 		// S6_OUT - PB14

6: TIM12_CH2 (PB15)
{ TIM12, GPIOB, Pin_15, TIM_Channel_2, TIM8_BRK_TIM12_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource15, GPIO_AF_TIM12},
`

From the code I believe (similar to the Revo) understand that they clash with TIM8.
I have therefore now added the Colibri to the Revo Timerclash definition as follows:

`
#if defined(REVO) || (COLIBRI)
            if (init->useMultiShot || init->useOneshot || isMotorBrushed(init->motorPwmRate)) {
                ppmAvoidPWMTimerClash(timerHardwarePtr, TIM12, init);
                ppmAvoidPWMTimerClash(timerHardwarePtr, TIM8, init);
            }
#endif
`
This has not helped with the motor output on these 2 channels though (guess this is only relevant for the PPM input.

Question on the motors: is there another output definition somewhere which needs to be adjusted to take these 2 special channels on TIM12 into account?

Thanks !

MJ666, 2016-04-09:
@IdefixRC after some more checking I cold not find a problem in the timer definition and assignments. :-(

IdefixRC, 2016-04-09:
Thanks for the help. 
Is it confirmed that the her mix works on the other targets? 
Could it have something to do with the UART and how it is defined in the target.h?

Thanks

MJ666, 2016-04-09:
Non of the usarts using pin PB14 and PB15. Please can you try this:

   { TIM3, GPIOB, Pin_0, TIM_Channel_3, TIM3_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource0, GPIO_AF_TIM3},    				// S1_OUT - PB0 
   { TIM3, GPIOB, Pin_4, TIM_Channel_1, TIM3_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource4, GPIO_AF_TIM3},    				// S2_OUT - PB4 
   { TIM3, GPIOB, Pin_1, TIM_Channel_4, TIM3_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource1, GPIO_AF_TIM3},    				// S3_OUT - PB1 
   { TIM1, GPIOB, Pin_15, TIM_Channel_2, TIM1_CC_IRQn, 0, GPIO_Mode_AF, GPIO_PinSource15, GPIO_AF_TIM1}, 		// S4_OUT - PB15 
   { TIM3, GPIOB, Pin_5, TIM_Channel_2, TIM3_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource5, GPIO_AF_TIM3},    				// S5_OUT - PB5 
   { TIM1, GPIOB, Pin_14, TIM_Channel_1, TIM1_CC_IRQn, 0, GPIO_Mode_AF, GPIO_PinSource14, GPIO_AF_TIM1}, 		// S6_OUT - PB14 
   { TIM10, GPIOB, Pin_8, TIM_Channel_1, TIM1_UP_TIM10_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource8, GPIO_AF_TIM10},    		// S7_OUT - PB8 
   { TIM11, GPIOB, Pin_9, TIM_Channel_1, TIM1_TRG_COM_TIM11_IRQn, 1, GPIO_Mode_AF, GPIO_PinSource9, GPIO_AF_TIM11},	// S8_OUT - PB9

IdefixRC, 2016-04-11:
@MJ666 unfortunately this did not solve the problem. Motor 4 and 6 are still not responding. :-(
I believe this is also correct (Quanton has TIM9 included here but not in the definition above. error?):

```
#define USED_TIMERS  ( TIM_N(1) | TIM_N(2) | TIM_N(3) | TIM_N(5) | TIM_N(8) | TIM_N(10) | TIM_N(11) | TIM_N(12))

#define TIMER_APB1_PERIPHERALS (RCC_APB1Periph_TIM2 | RCC_APB1Periph_TIM3 | RCC_APB1Periph_TIM5 | RCC_APB1Periph_TIM12 | RCC_AHB1Periph_GPIOA | RCC_AHB1Periph_GPIOB | RCC_AHB1Periph_GPIOC)
#define TIMER_APB2_PERIPHERALS (RCC_APB2Periph_TIM1 | RCC_APB2Periph_TIM8 | RCC_APB2Periph_TIM10 | RCC_APB2Periph_TIM11 )
```

Only other specific Colibri related code I could find which I'm not yet sure whether it is incorporated or not is in the old usb_bsp.c code related to the GPIO_Speed.
I can see that Revo here is at 100Mhz but Colibri was set to 25Mhz. Here is the original code:

```
#ifdef COLIBRI
  GPIO_InitStructure.GPIO_Pin = GPIO_Pin_9;
  GPIO_InitStructure.GPIO_Speed = GPIO_Speed_25MHz;
  GPIO_InitStructure.GPIO_Mode = GPIO_Mode_IN;
  GPIO_InitStructure.GPIO_OType = GPIO_OType_OD;
  GPIO_InitStructure.GPIO_PuPd = GPIO_PuPd_UP ;
  GPIO_Init(GPIOA, &GPIO_InitStructure);
#endif
#ifdef REVO
  GPIO_InitStructure.GPIO_Pin = GPIO_Pin_5;
  GPIO_InitStructure.GPIO_Speed = GPIO_Speed_100MHz;
  GPIO_InitStructure.GPIO_Mode = GPIO_Mode_IN;
  GPIO_InitStructure.GPIO_OType = GPIO_OType_PP;
  GPIO_InitStructure.GPIO_PuPd = GPIO_PuPd_NOPULL ;
  GPIO_Init(GPIOC, &GPIO_InitStructure);
#endif
```

I believe I have set the pin correctly in target.h but the speed??

IdefixRC, 2016-04-24:
@MJ666 any thoughts? Thanks

MJ666, 2016-04-24:
For the moment I don't have another idea. Since yor fc is working with an different firmware it is also not a problem with the hardware of your board.

IdefixRC, 2016-05-26:
I was busy with work and had no time to test this further. Having said that I definitely still want tonsee it happen. This weekend looks like I will find some time to test further. Fingers crossed

IdefixRC, 2016-05-27:
played around with the GPIO speed in usb_bsp.c as this was the only area I could find that still was somewhat different to sambas old code. No success.
I realized though that if USE_PWM_RATE is disabled in the gui that the motors do not work at all anymore (before and after GPIO speed ranges).

proggod, 2016-06-19:
Any developers willing to work on this contact me directlly :)

closed 2016-06-19 by proggod