rs2k/raceflight #128: Sbus and telemetry in single uart

carebear482, 2016-03-25:
I was wondering is it possible to make it work to use sbus on RX pin and telemetry on TX pin? Unfortunately revo nano is poor with uart ports.

blckmn, 2016-03-26:
this is something I will be implementing for the BlueJayF4. The Rev2 exposes the Receiver UARTs TX pin for this purpose - this can be adapted for other targets too where applicable. So watch this space.

proggod, 2016-06-19:
Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod