rs2k/raceflight #132: Revert to RC8

blckmn, 2016-04-04:
@MJ666 @rs2k 

I'm thinking to reset the main Raceflight branch to RC8, then rather than merge BF, I'll cherry pick each commit that will add value / provide fixes. The BF merge prematurely executed is not at all stable.

Thoughts?

MJ666, 2016-04-04:
@blckmn RC8 does not contain many other improvementside which are done before the large BF merge. This schould be merge back to this? May be we can roll back to the state before the BF merge.

blckmn, 2016-04-04:
Sounds good. I'll do it tonight, and then we can cherry pick the betaflight enhancements to maintain stability. There are some enhancements afterwards also that I'll replay too. 

Theres a considerable amount of work recently on CF configuration that I would like to cherry pick in too.

rs2k, 2016-04-05:
Yes, I think cherry picking is definitely the way to go. Large merges have never done well for me in the past. Another large issue is that the flight improvements done to the F1s and F3s never carry over well to the F4s. The F4s are flying incredibly well on RC8 at 8KHz +.

MJ666, 2016-04-05:
I think commits which affects the driver structure, io or configuration are relatively save to merge. If this is affecting the flight code we need to carefuly pick what is useful an test step by step. Actualy we are using an custom build based on RC8 (pwm changes and some AlienFlight specific upates) since this its working quite well. We never get an firmware build fom actual master to perform well on the AlienFlight F4 FC's. Looking forward to this changes.

rs2k, 2016-04-05:
The flight code in RC8 is specifically designed to work well on F4s. It's unfortunately we haven't been able to find a good solution that works for both F1/3s and F4s. I did spend a few weeks attempting to make all STM32 families happy, but in the end I went back to what worked best for F4s.

I believe this has to do with the way the MPUXXXX handles gyro divisions. If this is true, I would expect SPI F3s to fly well on RaceFlight as well. I'll need to test this theory.

blckmn, 2016-04-05:
https://github.com/rs2k/raceflight/pull/135

I would like to bring in 3D support, some bugs fixes, and perhaps the vbat compensation?

Any other suggestions?

blckmn, 2016-04-05:
I'll also add in the PWM rework.

MJ666, 2016-04-05:
There was quite some target specific updates wich we also should merge back in like for Quanton, AQ32, AlienFlight targets and likely much more. We should save the actual RaceFlight branch to pick commits as needed. I have not found out yet how cherry picking is working in eclipse. May be in the future it also makes sense the squash commits before merging them. This make the process more clean I think.

MJ666, 2016-04-05:
Looks the be this 2fe9681ab34fc81b7ce74b5279e0a48dc9894e74 is the last commit bevore the BF merge. I think it will be more easy to restart with this and apply the other commits wich are needed again than fixing issues at the actual branch?

blckmn, 2016-04-05:
https://github.com/rs2k/raceflight/pull/135 Reverts current branch to https://github.com/rs2k/raceflight/commit/2fe9681ab34fc81b7ce74b5279e0a48dc9894e74

It then has a few cherry picks... More cherries to apply though

blckmn, 2016-04-05:
@MJ666 I didn't want to force push or anything so I created a branch at the last stable commit ( the one you merged ) and then reset the index for the branch to Raceflight, leaving directory untouched. the differences were then committed. Putting it to exactly as it was before the merge, but leaving the previous commits in the history for cherry picking.

blckmn, 2016-04-07:
This has been done now :)

closed 2016-04-07 by blckmn