rs2k/raceflight #133: CC3D Revolution mini Pins

Jahnkeanater, 2016-04-05:
This isn't really an issue but here it is.

I have no idea how to figure out what the pins on the F4 board do So if somebody knows where I can find a schematic to figure out what pins do what that would be helpful.

As far as I can tell the revo mini is a full revo without the RF module. It has a STM32F405RGT6 processor with 64 pins. and it is very small. It also has a barometer and 3 axis magnetometer.
http://www.hobbyking.com/hobbyking/store/__98333__Mini_CC3D_Revolution_32bit_F4_Based_Flight_controller.html

I traced out all the connections and figured out where the pins connect to the STM32F405RGT6
starting at the top left going counter clockwise.

oplink air port
gnd, vcc, 6, 2, 3, 31, 4

Main port
gnd, vcc, 58, n/c

Flexi port
gnd, vcc, 45, 46

sonar/current port
gnd, vcc, 25, 57

Its strange that the main port has a pin that does not connect to the f4 at all?
Does this have something to do with the can bus?

There are also 4 test pads inside marked -, +, K, O.
The + pad has 3.3v and K connects to pin 1, O connects to pin 62.
I'm hoping these two pins are UART 6. This would make soldering a serial receiver fairly easy.

blckmn, 2016-04-07:
I don't have a Revo mini to confirm but there is a Revo mini target, and the pins are described in each target header now

Jahnkeanater, 2016-04-07:
There is a revo nano target. The revo mini board is a full revo without  the radio module. I'm running raceflight on it now without any problems. Everything I have checked has the same layout as the revo. The only difference the the weird oplink plug. I'm guessing those pins correlate to the pads on the bottom of the revo for the RF module.

blckmn, 2016-04-07:
Ah - ok. So is it all good for you?

blckmn, 2016-04-09:
@Jahnkeanater do you happen to know which pin the other side of UART1 inverter is connected to? for REVO fullsize it is PC0. I am assuming it is the same...

Jahnkeanater, 2016-04-10:
I believe its pin 60 on the f4 not sure if that's PC0? UART1 is working with the revo code so I'm assuming its exactly the same. The inverter is even working properly. I haven't tested UART6 because its mixed in with all the receiver pins.

mavericktw, 2016-04-12:
I am using this very small board called "CC3D Revolution Mini" with RC8 for REVO and main functions are no problem but buzzer and LED strip are not working...

CFBronson, 2016-05-08:
I have this version of the revo as well. I cannot figure out how to get raceflight on it. I seem to be able to get the raceflight configurator to recognize the board and gyro but for some reason RaceFlight gui knows the board isnt flashed raceflight fw. I cannot enable multishot or any of the other raceflight features. I assume I have a reading disability and that i have missed a detail or step in the process.. Anyone care to give a guy a hand? TYIA -Charles

proggod, 2016-05-08:
To get into DFU mode I think you need to do some delicate soldering, and connect some wires to some pins on the processor.  I trashed mine trying this :)

Although maybe there is a way to do it using the openpilot boot loader, hopefully someone else will speak up.

MJ666, 2016-05-08:
There must be an resistor pulling pin 64 (boot) down to ground. This should be more easy to reach than the pin itself. I have two of the Revo minis an need to take a look.

proggod, 2016-05-08:
Thats what rs2k thought too, i did a bunch of checking and couldn't find one, but if you find one let me know, maybe i'll buy one again.

CFBronson, 2016-05-08:
Hi guys...Let me start by saying I have watched all the videos i can find about flashing this FC with RF. Followed all the instructions i found online and installed VCP drivers. Im not sure where else to ask for help.. Ive been trying all day and Im not sure if im doing something wrong but i have the same board and I just cannot get raceflight to work. 
      It seems like i can get the bootloader to flash because I can open the RFgui and see that it recognizes my board movements but i cannot access any RF features,also the RX/TX is not responding on the graphs. Maybe its something I did wrong but I did change the RX input to ppm and saved and rebooted but nothing happens still.. TYIA 
      Another detail that may mean something.. It seems like everytime i save and reboot I have to manually stop the gui from connecting to COM1 and select the correct com port and select connect.

totaltmega, 2016-05-12:
There is two unsoldered pads(SBL) that you connect to each other to pull pin 60 to vcc. an easy way is to just touch the two pins at power up with a small metal object like a thin knife, then it stays in DFU mode until you remove power. Flashing with Revo target works fine but I'm having problems with PPM on releases newer than RC8, the signals are coming in as a much lower uS value, about 770uS to 1200uS...
[image]

totaltmega, 2016-05-12:
Also, I can probably help with testing on this board.

MJ666, 2016-05-14:
I flashed my Revo Mini's successfully with RaceFlight. This save me quite some time to find the BL pads. The build in LDO (TI REG113NA-3.3, marking R13C) will work up to 10V. This means 2S will directly work if the RX will be able to handle this also and nothing else is connected to the 5V power connections. Otherwise an BEC is needed. Another problem is 3.3v for an Spectrum Sat. Likely I will solder an cable to the + pad of the board.

Extent421, 2016-05-14:
Jahnkeanater are you sure the pin numbers you found are correct?  You have pin 60 as UART1 rx, but that's the boot pin.  pin 45 and 46 are PA12 and PA13 in the datasheet, which are UART1 RTS and JTMS-SWDIO, not UART3.  Maybe you're not counting from pin 1 on the package?

CFBronson, 2016-05-20:
> There is two unsoldered pads(SBL) The pads marked as circles, that you connect to each other to pull pin 60 to vcc. an easy way is to just touch the two pins at power up with a small metal object like a thin knife, then it stays in DFU mode until you remove power. Flashing with Revo target works fine except the erasing option.
 Thanks so much for this.. I finally got it.

> But I'm having problems with PPM on releases newer than RC8, the signals are coming in as a much lower uS value, about 770uS to 1200uS... 
 How does it function with Serial?

Extent421, 2016-05-21:
it flies great with serial.  I've been playing with an SD card on the air port as well and haven't run into any huge problems yet.

totaltmega, 2016-05-21:
@CFBronson 

> How does this affect flight with ppm?

I made a separate issue on the PPM problem #195 but basically it does not work at all.
I only have a PPM receiver so can not test serial unfortunately.

Best
Christoffer

CFBronson, 2016-05-29:
@totaltmega Hey bud, just wondering if you had a Dsmx sat to try out? Im not sure if it will help but just an idea that popped into my head.

totaltmega, 2016-05-29:
> @CFBronson 
@totaltmega Hey bud, just wondering if you had a Dsmx sat to try out? Im not sure if it will help but just an idea that popped into my head.

Sorry, I only have an Aurora 9 radio with OPTIMA 6 and 9 receivers, both flashed with custom fw to give out PPM (8ch) and PWM. PPM is working good on other FC's, and have had it checked out to PPM spec with an oscilloscope.

It is working on RC10 so something must have changed, is it easy to see what code was changed from RC10 to RC11?

Anything else I can help with?, really need this PPM to work with RF, soon done with my 220 build :)

Best
Christoffer

CFBronson, 2016-05-29:
I can assure you i have no idea how to check changes between releases. Ill give the sat a try in a few weeks when i start my 220x build. I have the sat running on my flexi port of my working craft at the moment. If it works with RF on the Revo Mini ill post up, if not ill be upgrading to an xjt module and x4r sb.

MJ666, 2016-05-29:
The PPM problem is likely related to the clock changes? https://github.com/rs2k/raceflight/commit/171a8d063d415fad270b13b65e829adad14c5c80

totaltmega, 2016-05-30:
> #if defined(STM32F40_41xxx)
-  uint32_t SystemCoreClock = 168000000;
+  uint32_t SystemCoreClock = 192000000;

yes probably, I'm happy to test if someone has any idea how to change the PPM code to the new frequency. Is there a VM available with the right build environment?

Best
Christoffer

#195

Minty81, 2016-06-06:
@totaltmega 
I've been unable to get my CC3D mini to go into bootloader mode by shorting those SBL pins.. Mine seems to get no lights on at all when those pins are shorted and 5V applied. I've got Openpilot BL 6 on it at the moment. Do you think that could have anything to do with it? Am i possibly shorting the pins too early? Do you power on the board then short the points right after power on? Thanks for your contributions already by the way!

totaltmega, 2016-06-06:
@Minty81 
I do it this way:
short the SBL pins while inserting it in the usb, with no external power.
This boots the board in DFU mode that should be ok until power off.
open RaceFlight GUI and see that it is in DFU mode and flash with "no reboot sequence" clicked in
and "Full chip erase" not clicked in.

If it doesn't work, then you can try to solder the pads together and then flash.

That you have OP in the board doesn't matter. Although you might not have the right drivers to see that it is in DFU?

Minty81, 2016-06-06:
@totaltmega 
I've tried just using the stm32102 drivers but no dice. The only way to get the board to connect to a virtual com port is through the librepilot CDC driver...then it connects as Revolution Virtual COM Port. Also..when I short the pins for flashing and apply power there are no lights...like it doesn't boot. Its the same board as your pics though. Got me scratching my head. Works with open pilot just fine! Hmm

totaltmega, 2016-06-06:
@Minty81
About no lights when in DFU mode seems ok, i remember that as well. So its probobly your drivers. 
On win 8/10, sometimes it is needed to install the driver in windows test mode to get unsigned drivers to be accepted. Cant see what drivers im using as im on a plane to spain.

Try to install these software packages http://www.st.com/content/st_com/en/products/development-tools/hardware-development-tools/development-tool-hardware-for-mcus/debug-hardware-for-mcus/debug-hardware-for-stm32-mcus/st-link-v2.html

proggod, 2016-06-06:
With the new zadig you don't need to install any drivers, check here for how to do it on revo: http://raceflightinfo.com/2016/06/01/flashing-cc3d-revolution-revo-with-raceflight/

Minty81, 2016-06-07:
@totaltmega @proggod 
Thanks for your help guys.  Spent hours on this.. soldered pinouts to the sbl and can confirm THE CC3D MINI DOES NOT HAVE LIGHTS WHEN IN DFU MODE. Sorry for the caps.. just making it easier for others who find this thread. Also....after many dozens of tries... my CC3D Revolution F4 Mini would only flash once 'No Reboot Sequence' was DISABLED....contrary to all the instructions I've read so far stating that it must be enabled if in bootloader mode... After DISABLING 'No reboot' I was able to flash and am up and running! 
Thanks again for your help guys. I hope my posts can help some other noob to get their CC3D Mini Revo flashed.  :dancers:

proggod, 2016-06-07:
@Minty81 I don't suppose you'd be willing to write an blog entry on our site raceflightinfo.net on how to do this?

Minty81, 2016-06-08:
@proggod Sure thing. Where abouts on the site do you want it?

proggod, 2016-06-08:
New topic for it.  You should be able to sign up and write articles and once you do just ask me to approve it, you can always email me at rf at prestongarrison dot com

Minty81, 2016-06-08:
@totaltmega  Would you mind if I borrowed your picture of the Mini and SBL pins for use in a how-to totalmega?

proggod, 2016-06-08:
@Minty81 i am guessing its fine to borrow them, they are on here :)

totaltmega, 2016-06-08:
@Minty81
Sure thing!

Jahnkeanater, 2016-06-20:
Yea I defiantly have The pins labeled wrong. I couldn't find the pin out for the F4 in any datasheet when I first got this board.