rs2k/raceflight #137: 3d defaults

Jahnkeanater, 2016-04-07:
When connected to a computer disarming goes to 1000 instead of center throttle.
This could be really dangerous. Maybe tell the escs to reset at disarm so they have to re initialize before being rearmed.

why is 3d_neutral not default at 1500?
new defaults should be
set 3d_deadband_low = 1406
set 3d_deadband_high = 1594
set 3d_neutral = 1500
set 3d_deadband_throttle = 20

Jahnkeanater, 2016-04-07:
Why would the behavior change depending if the flight controller is connected to the Configurator?

Jahnkeanater, 2016-04-07:
mwDisarm();
This function gets called a lot. where is it defined?

blckmn, 2016-04-07:
Good question I'll take a look

proggod, 2016-06-19:
3d support will be in coming betas, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod