rs2k/raceflight #139: Disable Motorstop when airmode is enabled.

Jahnkeanater, 2016-04-07:
In cleanflight airmode disables motorstop to prevent the motors shutting off when doing low throttle maneuvers. We should bring that feature to raceflight also.

blckmn, 2016-04-07:
Definitely. I'll look to bring that across this weekend.

blckmn, 2016-04-08:
what pull request in CF has this? 
https://github.com/cleanflight/cleanflight/pull/1766 ?

blckmn, 2016-04-09:
https://github.com/rs2k/raceflight/pull/141

blckmn, 2016-04-10:
@Jahnkeanater have created the pull request here: https://github.com/rs2k/raceflight/pull/141 needs to be tested.

blckmn, 2016-04-19:
@Jahnkeanater are you able to test?

Jahnkeanater, 2016-04-22:
Yea once I get my new escs I would be happy to test it. I don't know how to build the binary's from source but I could try to figure it out. Or if you sent me a hex file for a revo. I don't really know my way around github yet.

Jahnkeanater, 2016-04-22:
I'm guessing I would grab the source and swap out the 3 files you modified?

blckmn, 2016-04-22:
Here is a build of Next for the Revo. There may be other issues with the version, but let me know if it works ok for you.

https://www.dropbox.com/sh/py6f5bwyyuvcv0q/AACqTlBet-nQIxbd0jmz0Jhqa

Jahnkeanater, 2016-04-23:
Is this a special Airmode firmware or is this just RC11?

Jahnkeanater, 2016-04-23:
Seems to work. I haven't flown it yet but the motors seem to behave the way they should.

closed 2016-05-01 by Jahnkeanater