rs2k/raceflight #144: Revolution does not run on battery power

Doosa34, 2016-04-10:
When I plug in an external power source (battery), the FC does not boot (only the green light is on, there is no blinking).
When I connect my computer to the FC, it's okay, but if I connect meanwhile battery, then the connection is broken.

blckmn, 2016-04-10:
Hi there. This will be very specific to your setup / FC and unlikely to be related to RF.

But we will try to help. Are you using a 5v BEC? How's it all connected?

Doosa34, 2016-04-10:
I use 5V 3A PDB Bec, but I already tried with my ESC bec and the situation is the same. When I used the OpenPilot, it was not this issu, but I can imagine that I have a hardware something failure. :(

blckmn, 2016-04-19:
I am pretty sure this is a hardware issue specific to the board in question. I will close now, however if it is firmware - please advise and we will look into it.

closed 2016-04-19 by blckmn

Doosa34, 2016-04-19:
Problem solved! My GPS was connected in to the flexiport. After I disconnect the GPS and connect in to the main port, everything was good.
Thanks