rs2k/raceflight #145: Accelerometer loseing calebration mid flight.

Jahnkeanater, 2016-04-10:
I have had a reoccurring issue with auto level losing calibration mid flight. level becomes as much as 45 degrees away from true level. Something with horizon mode and lux pid controller must be changing the saved values for level. It seems like a sudden thing and is not a slow drift almost like calibration is being triggered mid air.

Maybe there is an issue with stick commands not being disabled when armed? I don't think I gave the accelerometer calibration command though.

Jahnkeanater, 2016-04-13:
Is anyone else having this issue with the revo code?

proggod, 2016-06-19:
Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod