rs2k/raceflight #146: Telemetry wont work on revo nano

carebear482, 2016-04-11:
As the title says or have anyone got this working?

carebear482, 2016-04-12:
It appears that telemetry part is working(shows in tasks list) but there is no built in inverter to handle s.port signal. Softserial could solve this issue.

carebear482, 2016-04-13:
Maybe it is possible to add few lines to revo nano target. I would need some help finding right timers and pins to make it happen, maybe something more. I just would like to get it done :)

#define USE_VCP
#define VBUS_SENSING_PIN PA9
#define USE_USART1
#define USART1_RX_PIN PB7
#define USART1_TX_PIN PB6
#define USE_USART2
#define USART2_RX_PIN PA3
#define USART2_TX_PIN PA2

#define USE_SOFTSERIAL1
#define USE_SOFTSERIAL2
#define SERIAL_PORT_COUNT 5

It would be nice if someone look into available timers and pins for that. Thats just raw copy from naze target.

#define SOFTSERIAL_1_TIMER TIM3
#define SOFTSERIAL_1_TIMER_RX_HARDWARE 4 // PWM 5
#define SOFTSERIAL_1_TIMER_TX_HARDWARE 5 // PWM 6
#define SOFTSERIAL_2_TIMER TIM3
#define SOFTSERIAL_2_TIMER_RX_HARDWARE 6 // PWM 7
#define SOFTSERIAL_2_TIMER_TX_HARDWARE 7 // PWM 8

rs2k, 2016-04-23:
telemetry inversion is only possible on soft serial. Soft serial is untested on the F4s, but there's plenty of CPU overhead to run upwards of 115K baud when only running at 8 KHz on the gyro.... assuming the task scheduler doesn't get in the way.

proggod, 2016-06-19:
Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod