rs2k/raceflight #148: Odd 3D Mode Behavior

MrKarver, 2016-04-13:
When armed at neutral stick (1500us) input the motors stop instead of idling in the direction the stick entered center. The bigger problem is when I disarm the motors go to Low idle (reverse direction idle) instead of motor stop. 

Oddly, if I disarm (so motors idling)  and then go to the motor tab I can see the FC outputting motor low idle command. I then arm and disarm while on the motor tab and then the FC goes to motor stop like it should 1500us and works as long as I am on the tab. So motor stop while disarmed works on the tab but idle while at center stick still does not work while on the motor tab.

My esc's are set to motor stop at 1500 and 3d_Neutral is set to 1500. My Low and high are also set correctly. 

Revo running Raceflight rev. 1 RC9
Blheli w/multishot 14.4.2_IM

Quad flies fine non 3D and for the most part 3D is working just idle and disarmed are backwards.

MrKarver, 2016-04-13:
Can anyone confirm or deny this?

blckmn, 2016-04-13:
Good analysis. I'll take a look tonight and let you know shortly

MrKarver, 2016-04-14:
I swapped back to an SP Racing F3 running Betaflight 2.6.1 and OneShot and it works as I would expect it. As it did when I tried it last in Betaflight 2.4.

MrKarver, 2016-04-24:
Did this get addressed in RC11? Should I try again?

MrKarver, 2016-05-16:
This works now!

proggod, 2016-06-19:
3d mode will be supported in new betas, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod