rs2k/raceflight #156: New FW still flip of death on front flip only

mattyra, 2016-04-20:
vo chaps...
Front flip still happening to flip of death see CLI Below
I have tried with out gyro sync on and in different accelerometer modes...
IT only happens when front flipping (RATES are LOW)
I have Tried in AIR MODE, ACRO + Mode and no mode and still it happens always in front flip..
Entering CLI Mode, type 'exit' to return, or 'help'
# dump
# version
# RaceFlight 16.04.17a - RaceFlight Release 1 RC9 - Gyro Overflow Fix /REVO Apr 17 2016 / 19:08:03 (6ae0fac)
# dump master
# mixer
mixer QUADX
mmix reset
smix reset
# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -MULTISHOT
feature -USE_PWM_RATE
feature -RESERVED
feature -TX_STYLE_EXPO
feature -SBUS_INVERTER
feature RX_SERIAL
feature SOFTSERIAL
feature FAILSAFE
feature BLACKBOX
feature MULTISHOT
feature USE_PWM_RATE
feature SBUS_INVERTER
# map
map AETR1234
# serial
serial 20 1 115200 57600 0 115200
serial 0 64 115200 57600 0 115200
serial 2 0 9600 57600 0 115200
serial 5 0 115200 57600 0 115200
set emf_avoidance = OFF
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set rssi_ppm_invert = OFF
set input_filtering_mode = OFF
set rc_smoothing = ON
set roll_yaw_cam_mix_degrees = 0
set min_throttle = 1065
set max_throttle = 1990
set min_command = 1000
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set enable_fast_pwm = OFF
set motor_pwm_rate = 16000
set servo_pwm_rate = 50
set disarm_kill_switch = ON
set auto_disarm_delay = 5
set small_angle = 25
set fixedwing_althold_dir = 1
set reboot_character = 82
set gps_provider = NMEA
set gps_sbas_mode = AUTO
set gps_auto_config = ON
set gps_auto_baud = OFF
set serialrx_provider = SBUS
set spektrum_sat_bind = 0
set telemetry_switch = OFF
set telemetry_inversion = OFF
set frsky_default_lattitude = 0.000
set frsky_default_longitude = 0.000
set frsky_coordinates_format = 0
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 0
set hott_alarm_sound_interval = 5
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = OFF
set current_meter_type = ADC
set align_gyro = DEFAULT
set align_acc = DEFAULT
set align_mag = DEFAULT
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 0
set max_angle_inclination = 700
set rf_loop_ctrl = M4
set moron_threshold = 16
set imu_dcm_kp = 2500
set imu_dcm_ki = 0
set yaw_control_direction = 1
set yaw_motor_direction = 1
set yaw_jump_prevention_limit = 200
set tri_unarmed_servo = ON
set servo_lowpass_freq = 400
set servo_lowpass_enable = OFF
set failsafe_delay = 10
set failsafe_off_delay = 10
set failsafe_throttle = 1000
set failsafe_kill_switch = OFF
set failsafe_throttle_low_delay = 100
set rx_min_usec = 885
set rx_max_usec = 2115
set max_aux_channels = 99
set acc_hardware = 0
set baro_hardware = 1
set mag_hardware = 1
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = SPIFLASH
set beeper_off_flags = 0
set magzero_x = 0
set magzero_y = 0
set magzero_z = 0
# rxfail
rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 a
rxfail 4 h
rxfail 5 h
rxfail 6 h
rxfail 7 h
rxfail 8 h
rxfail 9 h
rxfail 10 h
rxfail 11 h
rxfail 12 h
rxfail 13 h
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h
# dump profile
# profile
profile 0
# aux
aux 0 0 1 1700 2100
aux 1 27 1 1700 2100
aux 2 0 0 900 900
aux 3 0 0 900 900
aux 4 0 0 900 900
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900
# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0
# rxrange
rxrange 0 1000 2000
rxrange 1 1000 2000
rxrange 2 1000 2000
rxrange 3 1000 2000
# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1
set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83
set gps_wp_radius = 200
set nav_controls_heading = ON
set nav_speed_min = 100
set nav_speed_max = 300
set nav_slew_rate = 30
set gtune_loP_rll = 10
set gtune_loP_ptch = 10
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16
set alt_hold_deadband = 40
set alt_hold_fast_change = ON
set deadband = 0
set yaw_deadband = 0
set throttle_correction_value = 0
set throttle_correction_angle = 800
set default_rate_profile = 0
set gimbal_mode = NORMAL
set acc_lpf_hz = 20
set accxy_deadband = 40
set accz_deadband = 40
set accz_lpf_cutoff = 5.000
set acc_unarmedcal = ON
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf = 0.600
set baro_cf_vel = 0.985
set baro_cf_alt = 0.965
set mag_declination = 0
set pid_controller = LUX
set p_pitch = 65
set i_pitch = 100
set d_pitch = 22
set p_roll = 55
set i_roll = 80
set d_roll = 18
set p_yaw = 110
set i_yaw = 180
set d_yaw = 0
set p_pitchf = 6.500
set i_pitchf = 1.600
set d_pitchf = 0.140
set p_rollf = 6.500
set i_rollf = 1.600
set d_rollf = 0.140
set p_yawf = 9.300
set i_yawf = 1.750
set d_yawf = 0.000
set level_horizon = 3.000
set level_angle = 3.000
set sensitivity_horizon = 100
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 55
set i_level = 55
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1
set gyro_lpf_hz = 70
set dterm_lpf_hz = 70
set yaw_pterm_cut_hz = 30
# dump rates
# rateprofile
rateprofile 0
set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 40
set pitch_rate = 40
set yaw_rate = 35
set tpa_rate = 0
set tpa_breakpoint = 1500
set acro_plus_factor = 20
#

rs2k, 2016-04-21:
I've compiled a new fix for testing.

https://www.dropbox.com/sh/7ln7p97su5f89c8/AAAkJYmy9k0FiYjGobOWt_qma?dl=0

rs2k, 2016-04-22:
Resolved in RC11

closed 2016-04-22 by rs2k