rs2k/raceflight #157: Front Flip and back Flip rates and DPS different

mattyra, 2016-04-21:
Hello...

So i though i would run some field test on the DPS, and i have noticed that on RC6.2 the black flip and roll are slower to that of a front flip...

So i re tested on lateste version and the same applied, could this be leading to the death roll issues where it only happens in a front flip, maybe the front flip rates are running different to the rest... ??

rs2k, 2016-04-22:
I have not observed this issue. What rates are you using? What flight modes?

rs2k, 2016-04-22:
The death roll issue is resolved in RC 11

proggod, 2016-06-19:
Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod