rs2k/raceflight #162: F4 based FCs designed for Raceflight.

rs2k, 2016-04-24:
I want to make sure I've got my information straight on all the FCs designed for RaceFlight.

@blckmn I have the BlueJay F4 listed on the first thread of the RCG RaceFlight Post.
@kc10kevin I have the KKNGF4 listed on the first thread of the RCG RaceFlight Post.
@MJ666 I'm missing a link for the AlienFlightF4.

I don't think I'll be posting links to the cloned Revos or Sparky2s. The QC of clones worries me. Bad FCs will give the firmware a bad reputation. The Sparky 2 clones have a lot of problems for example. The Falcon F4 and legitimate Sparky 2s aren't designed for RaceFlight, but they could have a link if their QC is up to par..

blckmn, 2016-04-24:
looks good @rs2k. Perhaps the VRCORE?

nathantsoi, 2016-04-24:
I'm designing a new one. Yet to be named... :)

MJ666, 2016-04-24:
@rs2k for now the best link is www.alienflight.com. The AlienFlightF4 is not in production yet and we have not relesed as open source.

kc10kevin, 2016-04-24:
Thanks for the link!

closed 2016-06-19 by proggod