rs2k/raceflight #163: Features to drop

blckmn, 2016-04-24:
If RaceFlight foes F4 only - and racing only - I am curious as to what features should / could be removed that would simplify the maintenance of the code base?

rs2k, 2016-04-24:
There are a lot of vestigial limits in the code. This was originally designed to run on 8 bit hardware and it still shows. There are practically no RAM and FLASH limitations with even the weakest F4. We should look into the flash config. The smallest practical storage we can use is 16K. All profiles should be loaded into RAM. I'd like to see switch based profile switching on the fly.

nathantsoi, 2016-04-24:
not that anyone asked for my opinion here, so feel free to ignore, but i've been exploring the raceflight (and betaflight/cleanflight/baseflight) code for a while and it seems that a more modular structure would be the best of both worlds. easy maintenance + having only the features one wants. the dronin (openpilot) project is a good model, it would be rad if we had their build system (and hal) in the baseflight world

rs2k, 2016-04-24:
@nathantsoi Do you have any idea how similar Dronin is to OP's PiOS? Working with PiOS was not a fun experience.

nathantsoi, 2016-04-24:
good questions, i'm not really sure. my guess is that it's the same as op.

i was more thinking about how modules are total encapsulated, their entry and exit points are in the scheduler, defines are all nicely contained. e.g. there's an osd module that can easily be compiled in or left our depending on what you want, and it only takes a minimal (one maybe?) #define.

MJ666, 2016-04-24:
We have very much potential for lots of functionality on the F4 platform. I wuld not remove feature permanently. As @nathantsoi mentioned many featurs can be removed easily during compile time to make an racing only firmware but we have the option the bring the other and likly new functionality back in more easy.

blckmn, 2016-04-24:
@MJ666. I was more thinking support for winged flight etc, stuff that is just not multi rotor racing. Most of it will be broken anyway...

MJ666, 2016-04-24:
For sure fixed wing support can be dropped but i would keep Baro, Mag, Sonar and GPS support for the future.

nathantsoi, 2016-04-24:
agreed :)

Jahnkeanater, 2016-04-25:
I know softserial is still in the configurator. I'm not sure if the code still has it but I think we can drop it. We have 3 full uarts open and realistically that should be good enough. Not sure how many people are using the OLED and sonar features either. Maybe think about having a race version without gps features and a version more geared towards gps. I think most people looking for gps are going to run librepiliot though.

ted-rcnet, 2016-04-26:
the YoupiF4 is made for running without baro and mag :)
the gps is secondary ;)

ankjaers, 2016-04-27:
Isnt softserial used for smartport telemetry?

ted-rcnet, 2016-04-27:
I do not use smartport telemetry only osd

proggod, 2016-06-19:
Raceflight will support all boards with most features, f1, f3, f4...

Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod