rs2k/raceflight #167: LED Broken

proggod, 2016-04-27:
I am looking for something to work on in the code, and someone said LED is currently broken.   Does anyone have any details on what the issue is so I can get it fixed?

opiswahn, 2016-04-28:
It just doesn't work :) Code looks good I think, initialization is done properly, you can configure the led strip, the working code seems to run properly,  but there us no physical output to the strip it seems. 
I wasn't able to find it, but I got a feeling it has something to do with the dma controller and it's assignment.
Sorry but that's all I can give you.

opiswahn, 2016-04-28:
Tested it on sparky2, revolution and flip32,  all the same tesult

blckmn, 2016-04-28:
I think it might be also timing issues. I understand the ws2811/2 are very sensitive to timing... Currently waiting on some strips to arrive to test.

proggod, 2016-04-28:
I just need to get this thing to debug in Visual Studio and I'll see if i can figure it out, been using these strips for other things.