rs2k/raceflight #168: New Config Format

proggod, 2016-04-28:
I'm going to create a new config file format, since we are running on F4 boards mostly, we should no longer be limited on memory, so we can use a new config file format that is easier to upgrade and add variables.   After discussing with rs2k we decided a text based format would be the most readable and easiest to use, but welcome for comment.    Basically we will attempt to load config, if its not detected as our new text based format, we will load it into memory, convert it to new format and write it out, after this point, adding variables will be easy..  Format would be:

variable_name:data<lf>
variable_name:data<lf>
variable_name:data<lf>
variable_name:data<lf>
variable_name:data<lf>

What does everyone think?

rs2k, 2016-04-28:
We have vast amounts of flash and RAM to work with and this is basically how the RFBL's config works. I approve of it but would love to hear other opinions.

ted-rcnet, 2016-05-01:
if it brings a better reading
why not :)

IngoMaindorfer, 2016-05-27:
What about the good old INI file format? It will bring you a hierarchy thus helps for modularization. Do you familiar with SCPI? It an ASCII based format from HP for configuration of the good old Oszis and labor equipment. It works basically like this: for asking a value: MODULE:VALUE1 ? and for writing MODULE:VALUE1 777. If you do the modularization right, the config GUI can ask for all possible values and can parse it automatically.

opiswahn, 2016-05-27:
Why ini if you can do xml. Even in c a charm to process...

blckmn, 2016-05-27:
Why XML... Too verbose. JSON or even BSON. :)

MJ666, 2016-05-27:
Simple ASCII formats are more easy to manipulate. Should be somthing simple to use. Actualy I maintain an file with only the variables I have changed from default for each of my copters. It will be very easy to reconfigure them. Make an full CLI dump as many do for backup is likely not the best way. Loading this into an new build may complicate things.

proggod, 2016-05-27:
I love I put this issue in 29 days ago and now everyone is talking about it :)   Its going to be a simple text file format, because mj666 is right, its much easier to save and manipulate.   We are also moving to a better protocol for communication, one text based as well, and more expandable.

IngoMaindorfer, 2016-05-27:
Sorry for repeating me, before reinventing the wheel I kindly ask to have a look at this: https://en.wikipedia.org/wiki/Standard_Commands_for_Programmable_Instruments

closed 2016-06-19 by proggod