rs2k/raceflight #170: Always bind on startup mode w/ Spektrum

proggod, 2016-04-30:
Add ability to have spektrum always go into bind mode on startup.  This will allow you to always use same model for all quads, since settings are always the same, and use radios like the DXE without having to interface with your phone to switch models.  A lot of Eflite aircrafts work like this.

Jahnkeanater, 2016-05-01:
https://github.com/cleanflight/cleanflight/blob/master/docs/Spektrum%20bind.md
A feature like this already exists sort of...
It seems you have to define a bind pin when you build the firmware.
I think it would be fairly easy to remove the bind pin requirement and force bind every reset.

I'm assuming this code is also in raceflight.

proggod, 2016-05-01:
Ya i know bind through console exists, but when I do the new config changes, I'll make it so you can set it to always bind when battery is connected.  I more put this issue in to remind myself to add the feature.   Trying to add issues for things that either need to be fixed, or added, so I can organize what needs to be done :)

closed 2016-06-19 by proggod