rs2k/raceflight #173: Seperate TPA for yaw

ankjaers, 2016-05-02:
When tuning it's always yaw at full throttle that's giving me problems. Using rewrite I can run with almost 11.0 on P in the 0-70% throttle range (clean traces in BB + it feels great in the air), but to keep it from going nuts on full throttle I have to lower it all the way down to around 8. I feel like this could be fixed with seperate TPA for yaw.

rs2k, 2016-05-02:
I've had plans for a full TPS curve based on actuators with Kp, Ki and Kd separated for next. It's easy to make yaw separate from roll and pitch while doing that.

ankjaers, 2016-05-04:
Sounds good. I really think it will make tuning yaw easier.

ImbecileDK, 2016-05-04:
I would allso appreciate this seperate tpa for yaw too, yaw always seems to be a hazzle to tune while still feeling snappy

proggod, 2016-06-19:
Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod