rs2k/raceflight #175: Revo nano problems in rc12

carebear482, 2016-05-04:
Looptime went from 125 to 210 using default settings. Motor tab is very laggy and barely starts motors couldn't even calibrate escs.

carebear482, 2016-05-08:
Yesterday raceflight gui crashed and bricked my nano so deep that even zadig didn't find it. Also librepilot dfu rescue didn't work. Anyone have more ideas how to recover it?

proggod, 2016-06-19:
Please test betas and see if this issue is in any of them, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod