rs2k/raceflight #177: Lux Board Flashing

mattyra, 2016-05-05:
Hello Guys,

Can we look at the flashing of the Lux Board please, i have yet to flash one successfully it comes up as successful but the light stays blue and the board won't boot...

Matty

Clifftonlh, 2016-05-13:
I'm having the same problem with my Lux board..

krisquigley, 2016-05-13:
I am having the same issues too

slowGTO, 2016-05-20:
Same issue here as well on the Lux board.

TheDuke99, 2016-05-24:
Yep same issue here, flashes fine in Cleanflight but Raceflight flashes and then nothing, Windows10 says the USB device had an issue and was not connected...

mattyra, 2016-06-02:
@rs2k is the lux to be supported ??

proggod, 2016-06-02:
@mattyra yes it will be, we are having some issues with our new boot loader on it, but we will get them figured out..

mattyra, 2016-06-02:
Okay cool... Let me know if you need a test i have a few spare lux boards :-) @rs2k

proggod, 2016-06-19:
Lux will soon be supported, closing this issue since its based on an old version of our binaries, if you want access to betas please join our group at https://www.facebook.com/groups/1023175947742218/

Just trying to get github cleaned up so people can report issues with our betas....

closed 2016-06-19 by proggod

timdonovanuk, 2016-07-03:
Joined the Facebook group, don't see any posts about betas, asked about LUX FC, didn't get a response. Wouldn't it be better to use github for what is good at rather than using Facebook for code releases and issue tracking?