rs2k/raceflight #179: General Cleanup

rs2k, 2016-05-05:
The raceflight repo has been cleaned up.

The two major branches are "release" and "master". The old "raceflight" branch is now "betaflight_merge".

"master" is for stable, flight proven code and release is for new code. The "master" and "release" branches are based off of RC12. There are some serious issues with the most recent betaflight merge. Since there are going to be some pretty large changes coming to config, the drivers and task scheduler I believe the best solution is to base the newest code off of RC12. I will be looking at adding 1wire vcp support soon... after I resolve some of the bugs currently in RC12.

@blckmn @MJ666 @proggod @Jahnkeanater @nathantsoi @opiswahn

ted-rcnet, 2016-05-05:
Hi rs2k,
Fc youpif4 was ejected :/

rs2k, 2016-05-05:
@ted-rcnet Sorry about that, your code is still in betaflight_merge. You can submit a PR against release.

rodofpv, 2016-05-06:
So how about https://github.com/rs2k/raceflight/pull/176, should I recreate it for release and/or master branches?

proggod, 2016-05-06:
Yes can you do it against the release branch, and i can get it added.

ted-rcnet, 2016-05-06:
Ok rs2k no problem ;)

rodofpv, 2016-05-06:
Ok, @proggod

It is in https://github.com/rs2k/raceflight/pull/180

MJ666, 2016-05-08:
We have lost quite some valuable commits with this cleanup. This requires quite some work to get them back. For sure the latest code was not really flyable (at least not for us with the AlienFlight F4 boards). Actualy we are runnig an custom version based on RC9. Anyhow I see master an release already start to develop in different directions. I think we should define some more strict rules to go forward. This will avoid to require this kind of cleanups again.

closed 2016-06-19 by proggod