rs2k/raceflight #195: PPM problem on Revo Mini on all releases later than RC10

totaltmega, 2016-05-16:
Hi, my PPM signal does not come up in the GUI on all newer versions of raceflight later than RC10.
Im using the Revo Mini board from banggood, it is suppose to be compatible with the original Revo board.

Have the PPM pin moved since RC10?
Have the PPM logic changed from RC10 to RC11?

If you have instructions on how to build master I can see if the problem still exist.

Best
Christoffer

x-sergo, 2016-05-16:
Same problem here with Sparky 2.0. No PPM with my FrSky D4R-II receiver after flashing version higher than RC10.

Best,
Sergey

CFBronson, 2016-05-28:
I guess Ill have to upgrade my radio to use this FC with RF. >.>

blckmn, 2016-05-31:
No we should be able to get PPM to work again - it will just be some testing and clock adjustments.

x-sergo, 2016-06-04:
@blckmn you will also get it to work on the Sparky 2.0, right?

jtotham, 2016-06-16:
Any news on getting cPPM working on RF again? At the moment I'm having to use PWM to get signal into this board as the RX doesn't support serial.

CFBronson, 2016-06-28:
Hey guys, Has anyone had any luck with a Sat running through the main or flexi ports on the Revo mini?

roninnn, 2016-08-06:
Hey, I just got the revo (normal one) and ppm doesn't seem to work as well with 2016-04-24cREVO, with parallel_pwm works as expected

totaltmega, 2016-08-07:
Yes its the same issue, not even working on the latest betas available on slack.
I was told on slack https://raceflight.slack.com/messages/general/ that it is not prioritized due to that PPM is slower than serial input. 21ms (PPM 8ch) vs 7ms (SBUS) but PPM is the only thing I have so I´m stuck with RC10. If someone have an idea of what changed from RC10 to RC11, like clock settings, prescaler etc? I can help build the code and test it :)

blckmn, 2016-08-07:
This repo has limited development now with the move to the new code base. Im not sure when or if that's going to change. If you want to we can look at Revo nano ppm in Betaflight now that 3.0.0 is being released as RCs.

https://github.com/betaflight/betaflight

totaltmega, 2016-08-07:
Terriffic that betaflight now supports F4, Just tested it out to see if PPM
works on my Revo mini (REVO target) but it did not.
I tested both the pin 10 and pin 5 of the Flexi-IO port
http://opwiki.readthedocs.io/en/latest/user_manual/revo/revo.html
I dont know the PPM status on the nano tho.

I created an issue for betaflight
https://github.com/betaflight/betaflight/issues/915

Best
Christoffer

CFBronson, 2016-08-07:
:/ ouch  being "deprioritized" hurts.
Dumb question but shouldn't I be able to get a spektrum sat to work?
That's serial right? Even though the rx-tx link is dsmx the rx should be
speaking serial to the fc no?  -cheers

>
> Yes its the same issue, not even working on the latest betas available on
slack.
> I was told on slack https://raceflight.slack.com/messages/general/ that
it is not prioritized due to that PPM is slower than serial input. 21ms
(PPM 8ch) vs 7ms (SBUS) but PPM is the only thing I have so I´m stuck with
RC10. If someone have an idea of what changed from RC10 to RC11, like clock
settings, prescaler etc? I can help build the code and test it :)

totaltmega, 2016-08-19:
We have it working in betaflight now!
see https://github.com/betaflight/betaflight/issues/915

Please adopt the changes to raceflight if possible :)

Best
Christoffer