rs2k/raceflight #200: Private code access

rodofpv, 2016-05-25:
Is it possible to get access to that non-public branch/repo discussed below?

I tried to contact PrestonG about it, but didn't get any reply so I guess he doesn't read PMs on rcgroups.

I like to help with development when I have time, but when I know there's another codebase with a lot of changes I am afraid that the code in the public repo might be already outdated.

> Originally Posted by PrestonG
> > Originally Posted by rotceh_dnih  View Post
> > Thanks mate  that got me RC12 to work on , but say i wanted to work on this "NEXT" release you guy keep talking about , Is that on github ? i seen release branch was last updated 9 days ago is that the one? if its not public yet i can wait just if theres a chance of using it i'd much rather do that and get the speed advantages  
> 
> Is not public yet, we don't feel comfortable releasing it till we get it more solid and bug free. If you want to be able to test pre-releases you can always join the test team, and if you are a developer looking to help let us know and we can talk about that as well.

ted-rcnet, 2016-05-26:
same for me i like to help with development

blckmn, 2016-05-26:
I think it'll just be a matter of time. Hopefully we will see the changes here soon enough.

proggod, 2016-05-26:
We are working on it guys, the core is completely different, till this is finalized really its too hard to bring in outside help.  We are bringing everything modularized, so its going to eventually be easy for people to add boards and features.  we are getting rid of most of the ifdef code among other things.

IngoMaindorfer, 2016-05-27:
Modularization and get of #ifdefs sounds promising. Also like to step in to help. What do you think about support for newer MCUs like F7 based or Cortex A9 in bare metal mode?

ted-rcnet, 2016-05-27:
wath is the added value ?

IngoMaindorfer, 2016-05-27:
Nowadays you get for two bucks more a Cortex M7 over a M4. It's only a matter of time when these are lurking from the dark. Cleanflight has a real tight coupling to the used MCU because of intensive use of ST's std lib. But why not decoupling this a bit more and get more flexible for the future?

proggod, 2016-05-27:
Well if anyone has any good amount of time to help with coding message me on rcgroups, or facebook group or slack.  We are trying to bring more people into helping right now to push this out a bit faster, not quite ready to bring public yet though.

rodofpv, 2016-05-30:
I don't have much free time, but hopefully I can help with some shorter hack. Tried to contact PrestonG on rcgroups meanwhile. If it would help please feel free to add me to the slack, my mail is the same username as on github on gmail.com.

MJ666, 2016-06-01:
@proggod also send an PM via RCG regarding development help.

rodofpv, 2016-06-03:
Didn't get any response on rcgroups. Could you add me to the slack so that we can communicate easier? It is a bit frustrating to wait over a week for a response. (rodofpv gmail com)

proggod, 2016-06-03:
invite sent, rcgroups isn't the best way to communicate slack or facebook is better.. anyone else need an invite just let me know your emails

ted-rcnet, 2016-06-15:
is it possible to have more visibility on the progress RaceFlight because here it's going nothing ;)

proggod, 2016-06-15:
Join the facebook group if you haven't, we are putting out betas as we test them there.

ted-rcnet, 2016-06-15:
Could you add me my address is (ted rcnet.com)

proggod, 2016-06-16:
https://www.facebook.com/groups/1023175947742218/?ref=bookmarks

Just go there

proggod, 2016-06-19:
CLosing this issue, in future anyone that wants access to active beta development contact me directly, but we only want developers that have a good time to work on the project.  Otherwise wait till our source is more stable before we do any source releases.

closed 2016-06-19 by proggod

ted-rcnet, 2016-06-19:
Can I access the code Added YuPiF4 target ?
Pull request #198

proggod, 2016-06-19:
We will gladly move that into our new beta builds, but we need you to mail us out some boards for testing as well.   Can you contact me on slack/facebook/rcgroups..