rs2k/raceflight #225: License violation on binary release

mlyle, 2016-09-20:
@hydra @rs2k 

Despite claims of a "clean rewrite", the current code clearly infringes the GPL from Cleanflight and is obviously derived code.   Here's some disassembly of the current Revolution binary:

```
   21874:       041b            lsls    r3, r3, #16
   21876:       0c1b            lsrs    r3, r3, #16
   21878:       f043 0330       orr.w   r3, r3, #48     ; 0x30
   2187c:       8013            strh    r3, [r2, #0]
   2187e:       8813            ldrh    r3, [r2, #0]
   21880:       b29b            uxth    r3, r3
   21882:       f043 0340       orr.w   r3, r3, #64     ; 0x40
   21886:       8013            strh    r3, [r2, #0]
bit_h_reset
   21888:       2180            movs    r1, #128        ; 0x80
mpu_ra_pwr_mgmt_1
   2188a:       206b            movs    r0, #107        ; 0x6b
   2188c:       f007 ff44       bl      0x29718
atemptsRemaining = ... 21?
   21890:       f04f 0915       mov.w   r9, #21
   21894:       2096            movs    r0, #150        ; 0x96
delay 150
   21896:       f010 f989       bl      0x31bac
   2189a:       4632            mov     r2, r6
   2189c:       2101            movs    r1, #1
MPU_RA_WHO_AM_I
   2189e:       2075            movs    r0, #117        ; 0x75
   218a0:       f007 ff1a       bl      0x296d8
   218a4:       f89d 3064       ldrb.w  r3, [sp, #100]  ; 0x64
MPU6000_WHO_AM_I_CONST
   218a8:       2b68            cmp     r3, #104        ; 0x68
Subtract 1 from attempts remaining
   218aa:       f109 39ff       add.w   r9, r9, #4294967295     ; 0xffffffff
   218ae:       d025            beq.n   0x218fc
   218b0:       f019 09ff       ands.w  r9, r9, #255    ; 0xff
back to top of loop
   218b4:       d1ee            bne.n   0x21894
   218b6:       aa0e            add     r2, sp, #56     ; 0x38
   218b8:       9204            str     r2, [sp, #16]
   218ba:       aa0f            add     r2, sp, #60     ; 0x3c
   218bc:       f894 3230       ldrb.w  r3, [r4, #560]  ; 0x230
   218c0:       9202            str     r2, [sp, #8]
   218c2:       f7ff b967       b.w     0x20b94
   218c6:       460a            mov     r2, r1
   218c8:       9109            str     r1, [sp, #36]   ; 0x24
   218ca:       210c            movs    r1, #12
   218cc:       9105            str     r1, [sp, #20]
   218ce:       210b            movs    r1, #11
lots of stack manip
   218fc:       4632            mov     r2, r6
   218fe:       2101            movs    r1, #1
# MPU_RA_PRODUCT_ID
   21900:       200c            movs    r0, #12
   21902:       f007 fee9       bl      0x296d8
   21906:       f89d 3064       ldrb.w  r3, [sp, #100]  ; 0x64
# 0x14 to 0x18 is OK  (REVC4..REVD8)
   2190a:       2b14            cmp     r3, #20
   2190c:       d3d3            bcc.n   0x218b6
   2190e:       2b18            cmp     r3, #24
   21910:       d902            bls.n   0x21918
# 0x54 to 0x5a is OK.  (REVC4.. RevD10)
   21912:       3b54            subs    r3, #84 ; 0x54
   21914:       2b06            cmp     r3, #6
   21916:       d8ce            bhi.n   0x218b6
```

Vs. 

https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/accgyro_spi_mpu6000.c#L158

It goes on to lay out a very similar structure with function pointers followed by the gyro scale, etc.  There are many ways to do this, but the Raceflight code chooses to do the exact same operations in the exact same order with the exact same constants.

webbbn, 2016-09-20:
Coment #9 from [this issue](https://github.com/ShikOfTheRa/scarab-osd/issues/313) also raises questions about whether this was truly started as an independent work, or if it's a derivative product:

> ... We plan to replace all the MSP stuff in the next couple weeks with our new core, so it would be nice to know about any issues that make it less compatible.

There's also a serious question of how the developer could possibly be following "clean room" procedures when they clearly have intimate knowledge of the Baseflight/Cleanflight codebase.

http://digital-law-online.info/lpdi1.0/treatise27.html

mlyle, 2016-09-20:
> There's also a serious question of how the developer could possibly be following "clean room" procedures when they clearly have intimate knowledge of the Baseflight/Cleanflight codebase.

Yes, and you can't clean-room one piece at a time, reintegrating it with the main work, while giving it out to semi-public release and be OK.

Note that I didn't go searching all over the place for derivative code.  This is the first function I decompiled because I knew it would be easy to find.

digitalentity, 2016-09-20:
Also simply looking at the strings in the code you instantly notice mixer list that include HELI and AIRPLANE mixers, also you have strings for ancient sensors like BMP085 and ADXL345 (obviously for CLI but what's the point in implementing those if they use the fancy "super secret gyro"). Same for GPS-related PID names, same names for features in CLI, obviously reused blackbox etc.

I just opened two binaries side by side and strings look amazingly similar.

proggod, 2016-09-20:
Before this all gets blown out of proportion after @rs2k catches up on some sleep him and @hydra will both make a statement, talk about release time lines for our code, whats being released etc, before our actual release of our flight controller.   So stand by this week for details on that.  I can tell you we are releasing code, and we are integrating a lot of code into cleanflight over the course of the next few weeks, so we don't need to go on a witch hunt.   I can also tell you that we have an entirely new code base and core being released as well.   We are working together to make sure code bases play nice with everything, and cleanflight gets all the benefits of the raceflight development.  This has been planned for some time, and has not been made public.

closed 2016-09-20 by proggod

hydra, 2016-09-20:
The code is GPL, there's no arguing with that.  My feeling is that the raceflight guys should have kept their binaries private until they were ready to release the source as well and that was a mistake, one that I have communicated to them personally.  I too would like to be able to see full source for each release made via the raceflight slack channel which contains any of the Cleanflight codebase.

Anyone that has their code current in any version of raceflight that has been distributed in binary form up to now are of course entitled to do what they must if they are not happy.

That said I'm not going to burn any bridges as I'm looking forward to the future and to see how the raceflight guys can contribute to Cleanflight,  hopefully this situation will be resolved so that everyone is happy.

mlyle, 2016-09-20:
> Before this all gets blown out of proportion after @rs2k catches up on some sleep him and @hydra will both make a statement, talk about release time lines for our code, whats being released etc, before our actual release of our flight controller.

You're distributing binaries on the internet; you've got to ship source code at the same time, not when you feel like it.  GPL3 is quite clear on this.

At this point anyone who is a copyright holder in cleanflight can choose to DMCA you or litigate against you.  Just up to them whether they feel like it's worth it.

And beyond that-- how lame.  You mock people in your string table (@borisbstyle, @betaflight):

>  Sorry Boris, this ain't betaflight
>  Only original code allowed.

While being a bunch of plagiarists who aren't giving back.  You're happy to take everyone else's code for free, but will you share yours?  Naaaah.

tracernz, 2016-09-20:
> Anyone that has their code current in any version of raceflight that has been distributed in binary form 

There are a great many names on that list dating back to the beginning of MultiWii..

HABTomas, 2016-09-20:
Hi Guis
You are all doing great work with something that you love , work together have fun!!!  This shit is only hurting you and the hobby, it´s not fun at all! So thank to you all for creating something so great for all of us that don´t understand a shit about programing!!
So have fun!!

borisbstyle, 2016-09-20:
@proggod I wondered why I didnt get a reply on that CF issue last time when I mentioned rewrite :D. Just out of interrest.....why lying to everybody about the rewrite? Wasn't it just easier to write your own code when going to this direction? I mean with 25 year experience as code teacher you are capable of writing own code right?
I see you started selling own boards. I would expect you couldnt wait long enough to rewrite and decided to clone open source in the meanwhile for quicker time on the market in this one.

digitalentity, 2016-09-21:
@proggod, @rs2k do you think deleting comments is going to make you look better?

DieHertz, 2016-09-21:
@digitalentity This obviously belongs here:
https://en.wikipedia.org/wiki/Streisand_effect

glowtape, 2016-09-21:
Either way, you have one of the previous contributors clearly desiring release of the source code. Since you've admitted over at MWOSD that there's still at least their GPL code present in your codebase, a reference posted earlier which you've conveniently deleted, you're kind of bound into complying. So how about it?

WillManidis, 2016-09-21:
Why was the issue closed? This is clearly unsolved.

mlyle, 2016-09-22:
> Before this all gets blown out of proportion after @rs2k catches up on some sleep him and @hydra will both make a statement, talk about release time lines for our code, whats being released etc

We're all still waiting.

joshuabardwell, 2016-09-22:
I have an announcement, and @proggod has asked me to make it here first.

@proggod has been looking for someone that both the RF devs and the community can trust to do an audit of the RF code, to help answer the question of whether a GPL violation is occurring. The auditor will receive full access to the RF github repo, so he has to be someone who both the RF devs and the community can trust. 

The person that has been settled on is Jonathan Davis, Skitzo. 

When he's not being a Master of Flow, Skitzo is a game developer, so he's got the programming chops for the job. We all know that Skitzo switched over to RF, but he's not involved in the development of RF, and as far as I can tell, he's a straight shooter. For the record, @proggod asked me to do the audit, specifically because I am a big fan of Betaflight and not associated with the RF project in any way, but I don't feel that I am a skilled enough programmer to perform the audit.

The reason I'm making the announcement and not @proggod or @rs2k is because @proggod asked me to interview Jonathan for the "big reveal" of his findings and to help disseminate information about the audit through my public channels. Rather than Jonathan reporting to @proggod and maybe the info getting tainted or shaped, Jonathan will report to me directly, and @proggod and all the RF devs will hear the results on my channel, the same as everybody else.

This information isn't secret--in fact, it's public--but I would love to get any feedback that y'all have here before I make a wider announcement. What do you think of this idea?

joshuabardwell, 2016-09-22:
PS: During the announcement interview, I would also like to include someone who is an authority on the GPL, so that the right questions can be asked, to determine if the project is violating the GPL. If any of you have any suggestions about who that might be, contact me with them. Obviously, an air of impartiality is important...

digitalentity, 2016-09-22:
With all due respect, I can't say I agree.

Skitzo is flying RF and thus wants that firmware to continue, therefore he's not unbiased exactly as as any of RF/CF/BF/etc devs. 

For a trustworthy audit we need somebody completely independent and not interested in either CF or RF.

mlyle, 2016-09-22:
@joshuabardwell  --- very clearly a GPL violation is occurring.  The first function I decompiled/reversed is clearly derived from Cleanflight.  The exact same delay intervals, order to check registers, acceptable MPU version IDs are the same range; the structure filled with constants about scale are used.  The exact same floating point constants are used and not their reciprocals or other valid choices that would result in the same behavior.  They're placed into similar structures and not other layouts of memory.

There are thousands of possible design choices on how to do this, and somehow Raceflight has come to the exact same choices-- when written by developers that have access to the Cleanflight sources and experience working on that source code base.  This makes it very clear that Raceflight is derived from Cleanflight.

For reference, I am not involved in Cleanflight or Raceflight; I am a developer of dRonin-- and I have the utmost respect for the work Betaflight and Cleanflight are doing.

mlyle, 2016-09-22:
Also I would just like to state that I am disturbed that there is a seeming-attempt to be transparent here when comments from leaders of other related projects have been deleted by Raceflight maintainers in this very thread.  That does not bode well for the trustworthiness of this process.

tracernz, 2016-09-22:
So the purpose of the audit is to determine the extent of the violation, because that seems kind of irrelevant? :/

digitalentity, 2016-09-22:
@rs2k @proggod 
How about releasing the source? If it's 100% your authentic work you can chose to release it under a license that it prohibiting derivatives of the code (in a sense not an open-source license) - you'll be able to enforce that in court if you have to. 
But if you have some code derived from CF/BF/whateferFlight this be what GPL demands.

joshuabardwell, 2016-09-22:
@mlyle I'm not really the one you need to be arguing with. I'm neither a good enough coder, nor knowledgeable enough about the GPL to say whether or not a violation is occurring. I'm just providing a venue for findings to be announced, and hopefully to generate some discussion of the findings. If you are an expert on the GPL and can make a convincing argument that a violation is occurring, let's talk about you making that argument on my channel, where it can be heard.

digitalentity, 2016-09-22:
PS: You can publish only hardware drivers, R/C RX code, MSP code, CLI code (those are 99% unrelated to flight performance but they are a major (really major) part of the code.

digitalentity, 2016-09-22:
@joshuabardwell, I think @mlyle got a crystal clear argument in the first post of the thread. The only thing that was required from RF to prove him wrong was to throw us a bone - give us the source code of that driver so we could see for ourselves that it's not derived from CF. Apparently at the moment this won't be enough...

joshuabardwell, 2016-09-22:
BTW, for the record, I have a Bachelor's in CS from GA Tech, and I have been a hobby programmer basically my whole adult life. So it's not like I'm oblivious here, but the auditor has to be at least as good as the people he's auditing, and I'm not. I also know a bit about the GPL, but the intricacies of exactly what might constitute a violation, I don't feel qualified to judge in a case like this even if to others, it's cut-and-dried. The intensity with which some people feel that it's cut-and-dried actually makes me retreat more into a question-asking mode rather than a statement-making mode.

seaweeduk, 2016-09-22:
Have to say I agree with others that Skitzo isn't a particularly unbiased source for auditing this seeing as he's already flying it and has a vested interest in it. I'd also like to see some more information regarding his credentials as a programmer.

tracernz, 2016-09-22:
@rs2k @proggod You guys should probably have a good read (or three) of https://www.softwarefreedom.org/resources/2008/compliance-guide.html. You will find there are many things you didn't know. Section 5.2 (which presumably you are currently in violation of since you haven't ceased distributing the derivative work) should be now of particular interest. :wink:

borisbstyle, 2016-09-22:
Not much point of posting as all my messages get deleted, but someone will read it probably :D.

Who ever reviews. Don't forget to go historically though every released binary as if this would be the case it is important to know how many times GPL was violated.

dexX7, 2016-09-22:
@borisbstyle: users subscribed to this thread should receive new messages with email, so even if messages are deleted, they are certainly not lost.

seaweeduk, 2016-09-22:
Can confirm I see Boris' posts in my email inbox, it's disgusting that they are being deleted this whole thing is coming across as very shady indeed.

digitalentity, 2016-09-22:
Heh, and another message from Boris they removed. 
[image]

glowtape, 2016-09-22:
This is a farce beyond imagination. The first reply to this issue links to some other Github project where one of the owners of the Raceflight project admitted that there is still code from another GPL project in Raceflight at the time he made that comment. Up until that point there were beta releases quasi-publicly available. There's nothing to discuss or review here. Up to at least that date of comment, the source code to the relevant beta releases has to be made public.

@digitalentity There isn't selective releasing of source code. Any bit of GPL3 code in your project requires you to release ALL of it.

digitalentity, 2016-09-22:
@glowtape 
The latest binary release of RF has a date tag of "Sep 22 2016", however the last commit to this repository was made on  Jun 14 by @proggod. So where's the code of the last released version?

digitalentity, 2016-09-22:
@glowtape I must have misunderstood you, sorry. I agree, all GPLed code should be released. I just say if they did a rewrite they can at least give us samples of the code that are not affecting flight performance so we could compare and see for ourselves.

seaweeduk, 2016-09-22:
Here's the first post from Boris they deleted too, for anyone who missed it

[image]

mashu, 2016-09-22:
So Skitzo after whom flight mode is called is going to be  unbiased external auditor who is clearly going to bend reality for us. Us is those who dont live in this another dimension where clear GPL3 violations are debated ? This is getting more absurd every minute;)

joshuabardwell, 2016-09-22:
@glowtape maybe this is obvious to people with more experience with the GPL. I'm still in the "coming up to speed" process on the facts of this issue. If you or anybody else has facts that you think need to be a part of the story, please get in touch with me.

digitalentity, 2016-09-22:
Joshua, I think @glowtape is trying to say that Skitzo as an auditor is
going to be biased and his report won't prove anything.

Violating GPL is using the code to make derivative work and not
distributing the source along with it. There is evidence that RF might have
used the CLI,  Blackbox and gyro driver from Cleanflight, therefore they
are obliged to publish the whole code as GPL doesn't allow to combine open
source and proprietary code.

joshuabardwell, 2016-09-22:
@digitalentity I feel like one of the things that I contribute to the community is to translate complicated technical topics for general consumption. It feels like that ability might apply to this situation. Can we talk about this more, one-on-one?

digitalentity, 2016-09-22:
Can you email me your Skype (my email is in Github profile). I'll be online
in about 20 minutes.

glowtape, 2016-09-22:
I haven't actually said anything in regards of Skitzo.

What I however am saying is that there's more than enough pointing towards a violation that pretending that things are more intricate is a little disingenuous.

glowtape, 2016-09-22:
I mean, what is there even to discuss? @mlyle showed strong evidence that an old GPL'ed function is still present and @proggod of Raceflight itself said over on the MWOSD project that the MSP code still has to be replaced by their own implementation. All while there are public binary releases out.

mlyle, 2016-09-22:
And the saga continues in #226, where we pick out data structures from the Blackbox implementation and find them to be curiously similar ;)

OhmEye, 2016-09-22:
This is a ridiculous situation. Disputing that RF contains GPL'd code is nonsensical. It's proven, it's obvious, and it's logical. It's not a reasonable claim that the entire body of GPL'd code in RF has been rewritten from the ground up since June do be entirely original and non-derivative. The current binaries show beyond doubt that this has not happened, which is reasonable and obvious. It defies belief that it's possible for any audit of the source can result in a verification that the source is completely original from the June RF GPL'd codebase yet just happens to generate the same binaries.

MAVProxyUser, 2016-09-22:
Heh Kalyn played with fire and got burned? I'm not surprised. Reap what you sow...

joshuabardwell, 2016-09-22:
@OhmEye it's actually because the RF devs claims are so spectacular that I'm intrigued by the offer to audit. If a magician puts a rabbit in a box, and then bets you $100 that the box is empty... well, for one thing, you're a sucker if you take the bet, but at the same time, you've got to wonder where he hid the rabbit.

OhmEye, 2016-09-22:
@joshuabardwell Oh, I don't think anybody should turn down an offer to view their source. I also don't think one has to be a lawyer or GPL expert to compare RF source to BF/CF and see similarities. Of course I haven't seen their current source, but the idea that they stopped updating the github repo in June and wrote an entirely new project from scratch since that time with the same functionality and features without any code duplication or derivation is literally incredible. The fact that the binary contains the same code proves beyond doubt that some source is the same, even should there turn out to be source obfuscation, which on the face of it makes an audit unnecessary for the purposes of determining a GPL violation. GPL violation has already been proved, I don't see how it can be honestly disputed.

Meanwhile, if an audit is going to happen it should be by somebody acceptable to people on both sides of the claims, which typically means somebody accepted as unbiased. I think the purpose of an audit at this point would basically be to determine the extent of violation, how much GPL or derived code exists. I'm not sure why that matters other than for curiosity, since any at all is enough for violation and it's unreasonable in the extreme that they really have only some small overlooked things.

joshuabardwell, 2016-09-22:
@OhmEye from a legal perspective, I doubt any of this matters, because I doubt any stakeholders with a claim will reach the point of litigation.

From an ethical perspective, I definitely think there is a difference between "one line of code" which is technically a GPL violation, and wholesale copying.

In other words, as far as most of us are concerned, the extent of the violation is probably what we care about most.

mlyle, 2016-09-22:
@joshuabardwell now I'm up to 3 functions reverse engineered (out of 3 attempted), all identical to Cleanflight except one constant being different (21 attempts to initialize the gyro instead of 5).  The other issue has annotated assembly dumps and side-by-side code comparisons.  I can keep doin' this all day :P

OhmEye, 2016-09-22:
@joshuabardwell I also doubt this will actually go to litigation, and I think it's unlikely to capture the  attention of the EFF just based on the small userbase.

I agree there's an ethical difference between some inadvertent bits here and there and wholesale copying. I'm having a hard time understanding how anybody familiar with the history of the *flight codebase can suspend disbelief enough to entertain the notion that there is not a significant amount of copying or derivation in this case. Really, this is so very clearly not a situation where only a minor or overlooked amount of code was carried forward.

So yes, it's precisely because most of us are concerned about the extent of the violation that I think it's so egregious. Bluntly, my personal curiosity is more about discovering if there is any significant amount non-derived code at all. Yes, I suspect it's that bad.

webbbn, 2016-09-22:
@joshuabardwell Did you switch sides?  I thought you were here representing RaceFlight.  Are you now admitting that you believe their is a violation, but they don't plan to do anything about it?  Why don't the RaceFlight principals come here to defend themselves?

Even though we all have our individual projects, the open source flight controller community is pretty well connected, which is demonstrated by the players involved in this discussion, which are not all MultiWii/XFlight developers.  Does RaceFlight want to start a business based on theft and deception?

Reputation means something, and if RaceFlight thinks they have something unique that can't be duplicated and surpassed honestly by this community, then I think they might end up learning a hard lesson.

tracernz, 2016-09-22:
> Even though we all have our individual projects, the open source flight controller community is pretty well connected, which is demonstrated by the players involved in this discussion, which are not all MultiWii/XFlight developers. 

We just need some APM guys in here and we'd have everybody. I suspect they have enough of their own licensing issues to deal with at the moment with DroneCode though. :wink:

vadav, 2016-09-22:
@joshuabardwell In regards to using Skitzo as a neutral audit..  forget it..  he announced alread a month ago to working with the RF team..     https://youtu.be/PV-k1-yRgbg?t=324

mlyle, 2016-09-22:
Let's talk about the "Super Secret Gyro"!  Thanks to people who pointed me to the Raceflight beta code for the revolt board.

Let's look at what the gyro readout code looks like for the "Super Secret Gyro"!  Again, despite doing a "clean rewrite", they seem to have chosen the exact same structure layouts and order of function parameters as CleanFlight!

```
; save stack
   366b8:       b5f0            push    {r4, r5, r6, r7, lr}
; r3 = 0x20006acc (ram address) -- mpuConfiguration
   366ba:       4b69            ldr     r3, [pc, #420]  ; (0x36860)
; grow the stack 12
   366bc:       b083            sub     sp, #12

; let's put our argument out of the way for later use
   366be:       4604            mov     r4, r0

; pull out the word (function pointer) 4 bytes into this structure  -- mpuConfiguration.read()
   366c0:       685d            ldr     r5, [r3, #4]
; pull out the byte at the beginning of this structure (mpuConfiguration.gyroReadXRegister)
   366c2:       7818            ldrb    r0, [r3, #0]
; argument 2 =  where we grew the stack to (&data)
   366c4:       466a            mov     r2, sp
; arg1 = 6
   366c6:       2106            movs    r1, #6
; ret = mpuConfiguration.read(mpuConfiguration.gyroReadXRegister, 6, data);
   366c8:       47a8            blx     r5
; check for 0 retval, go to end / return if so
   366ca:       2800            cmp     r0, #0
   366cc:       f000 8086       beq.w   0x367dc

; very crude state machine math and floating point readout code follows
   366d0:       4b64            ldr     r3, [pc, #400]  ; (0x36864)
   366d2:       781a            ldrb    r2, [r3, #0]
; if r2 = 3 then r2 = 0
   366d4:       2a03            cmp     r2, #3
   366d6:       bf04            itt     eq
   366d8:       2200            moveq   r2, #0
; store the altered r2 back to the beginning of the structure.
   366da:       701a            strbeq  r2, [r3, #0]
```

vs. pios_mpu in cleanflight:

```
bool mpuGyroRead(int16_t *gyroADC)
{
    uint8_t data[6];

    bool ack = mpuConfiguration.read(mpuConfiguration.gyroReadXRegister, 6, data);
    if (!ack) {
        return false;
    }
```

rs2k, 2016-09-22:
Come on now, this is just getting pathettic. How many ways are there to read a gyro or print a string?

mlyle, 2016-09-22:
You have a configuration structure in memory that consists of a 1 byte register number, 3 bytes of alignment gap, and then a function pointer to do the read.  It is the exact same code structure as cleanflight.

The function pointer receives 3 arguments in this order-- the register number, the number of bytes to read, and a destination pointer.  It returns 0 on failure, which most c code doesn't do.  It's the exact same structure as cleanflight.

The target for the read is on stack.

There's dozens of choices of how to do these things.

> Come on now, this is just getting pathettic. How many ways are there to read a gyro or print a string?

I don't know--- millions or billions of equivalent, likely programs?  But you guys ended up with the exact same one.

treken, 2016-09-22:
As a former developer I can understand that when someone kangs your code and takes credit for it it's like a slap in the face. I won't lie, although I used another's code in many instances I would always give credit even if the code evolved into completely different strings/structure we would still preserve credit from the coder that the code originated from. Not even getting into GPL it's just an ethical issue above all. Someone will always stand on the shoulders of giants at some point

mlyle, 2016-09-22:
Let's pick something else then.  Cleanflight uses a somewhat improbable constant.

```
    return lrintf(throttle_correction_value * sin_approx(angle / (900.0f * M_PIf / 2.0f)));

which equals

3.14159265358979323846 * 900 / 2.0 = 1413.71669411540695730700

0x44b0b6ef in hex
```

Well look, there it is in the revolt image:

```
00024290  ef b6 b0 44 74 88 00 20  e4 86 00 20 64 91 00 20  |...Dt.. ... d.. |
```

Let's look at how address 24290 is used in the code:
```
indirected load of pointer from ram
   2402e:       4b97            ldr     r3, [pc, #604]  ; (0x2428c)
cr7 = 1413.71..
   24030:       ed9f 7a97       ldc     10, cr7, [pc, #604]     ; 0x24290
   24034:       edd3 7a00       ldcl    10, cr7, [r3]
floating point, what a pain.  multiply by throttle scale angle
   24038:       ee20 0a27       cdp     10, 2, cr0, cr0, cr7, {1}
   2403c:       eebd 0ac0       cdp     10, 11, cr0, cr13, cr0, {6}
   24040:       ee10 3a10       mrc     10, 0, r3, cr0, cr0, {0}
compare angle to 900
   24044:       f5b3 7f61       cmp.w   r3, #900        ; 0x384
   24048:       bfa8            it      ge
    if (angle > 900)
        angle = 900;
   2404a:       f44f 7361       movge.w r3, #900        ; 0x384

   2404e:       ee00 3a10       mcr     10, 0, r3, cr0, cr0, {0}
; divide by this 900 * m_PI / 2.0f constant
   24052:       eef8 7ac0       cdp     10, 15, cr7, cr8, cr0, {6}
   24056:       ee87 0a87       cdp     10, 8, cr0, cr7, cr7, {4}
; and truncate it
   2405a:       f00d f975       bl      0x31348
```

Compare to the cleanflight throttle angle correction code:  https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/imu.c#L495

```
    int angle = lrintf(acos_approx(rMat[2][2]) * throttleAngleScale);
    if (angle > 900)
        angle = 900;
    return lrintf(throttle_correction_value * sin_approx(angle / (900.0f * M_PIf / 2.0f)));
}
```

Miraculously, you guys picked the same values for cutoff angle and chose to do the math in the exact same ways.  You didn't convert to radians earlier in the process, you don't do a tangent correction; you choose this scaling instead of power scaling.  Very nifty-- it's like you are perfectly mentally attuned to make the same choices that the MultiWii-derived projects do.

Never mind that every other flight stack on the planet is different-- APM, PX4, OpenPilot et al, AQ32-- you just always happen to pick the constants, parameter conventions, units, math scaling that Cleanflight does.

glowtape, 2016-09-22:
Come on now, this is just getting pathettic. How many ways are there to chose an arbitrary cutoff angle?

tracernz, 2016-09-23:
@rs2k I will ask one simple, straightforward question which you should be able to answer easily. Are the binaries at http://raceflight.net/beta compiled from any GPL-licensed source code for which you (or other raceflight devs) are not the copyright holder?

[19 more replies]