rs2k/raceflight #226: More GPL violation in Raceflight - Altitude Hold

mlyle, 2016-09-22:
Here's a short introduction to reverse engineering.  Let's go looking for another constant.  How about blackbox, since it's been mentioned.

```
H Product:Blackbox flight data recorder by Nicholas Sherlock
```

Well, that was easy!... wait, maybe not.  Maybe just the header is required by some tools and it was decided to duplicate it verbatim here.

The GPL'd blackbox implementation in Cleanflight has some large tables and is data driven.  You could form these tables and structure members in many ways; or you could omit them and do it in code instead of making it data driven, etc.  There are a *lot* of possible choices.

```
    {"axisP",       0, SIGNED,   .Ipredict = PREDICT(0),       .Iencode = ENCODING(SIGNED_VB),   .Ppredict = PREDICT(PREVIOUS),      .Pencode = ENCODING(SIGNED_VB), CONDITION(ALWAYS)},
    {"axisP",       1, SIGNED,   .Ipredict = PREDICT(0),       .Iencode = ENCODING(SIGNED_VB),   .Ppredict = PREDICT(PREVIOUS),      .Pencode = ENCODING(SIGNED_VB), CONDITION(ALWAYS)},
    {"axisP",       2, SIGNED,   .Ipredict = PREDICT(0),       .Iencode = ENCODING(SIGNED_VB),   .Ppredict = PREDICT(PREVIOUS),      .Pencode = ENCODING(SIGNED_VB), CONDITION(ALWAYS)},
```

So let's look for this.  Basically, this is a structure that consists of a 4 byte pointer, and then 7 1 byte values.  Because of alignment, that'll make it 12 bytes long.  We expect to find a pointer to the string "axisP", followed by the value 1, etc.  

So first, axisP:

```
0003ffc0  6e 00 00 00 74 69 6d 65  00 00 00 00 61 78 69 73  |n...time....axis|
0003ffd0  50 00 00 00 61 78 69 73  49 00 00 00 61 78 69 73  |P...axisI...axis|
```
That looks like 0x3ffcc, or 0x0803ffcc once we convert it to a flash address.

Note that the addresses are in the opposite order here because the code is little endian.

```
0003e240  02 00 00 00 cc ff 03 08  00 01 00 00 01 00 00 00  |................|
0003e250  cc ff 03 08 01 01 00 00  01 00 00 00 cc ff 03 08  |................|
0003e260  02 01 00 00 01 00 00 00  d4 ff 03 08 00 01 00 00  |................|
```

So at 0x3e240, there's the pointer to axisP, followed by the number 0, 7 more bytes of stuff; then 12 bytes later axisP followed by 1, then 12 bytes later axisP followed by 2.  So out of many possible choices, Raceflight coincidentally arrived at the same overall code structure and data structure layout at the GPL'd implementation in Cleanflight.  The fields are in the same order, and are broken up into sub-datastructures for GPS, slow fields, etc.

From the addresses of these structures, it's trivial to locate the code using them and see the code is identical to the code in Cleanflight.  It's left as an exercise to the reader-- or a future episode of Finding-GPL-Violations-By-Reversing-Raceflight.

Tagging @thenickdude, because it's mostly your code.  :D

mlyle, 2016-09-22:
Next up, let's look at blackboxWriteSysInfo.

```
cleanflight:    if (blackboxDeviceReserveBufferSpace(64) != BLACKBOX_RESERVE_SUCCESS) {
raceflight:
   23e4c:       2040            movs    r0, #64 ; 0x40
   23e4e:       f010 fcd3       bl      0x347f8
   23e52:       2800            cmp     r0, #0
   23e54:       f47f af36       bne.w   0x23cc4

cleanflight: 
    switch (xmitState.headerIndex) {

raceflight:
; load xmitState
   23e58:       4db5            ldr     r5, [pc, #724]  ; (0x24130)
; load the first 32 bit word in it, which is headerIndex in CleanFlight
   23e5a:       682b            ldr     r3, [r5, #0]
; make sure the value is less than or equal to 12
   23e5c:       2b0c            cmp     r3, #12
   23e5e:       f200 82ba       bhi.w   0x243d6
; jump table / switch statement --23E6 table branch halfword. 
   23e62:       e8df f013       tbh     [pc, r3, lsl #1]
; Let's look at case 3, because it's most interesting.
   23e6c:       0310 
; OK.  23e62 + 4 (the next instruction + 0x0310 * 2= 24486  Did we do the math right? What's there?
   24486:       9916            ldr     r1, [sp, #88]   ; 0x58
   24488:       4820            ldr     r0, [pc, #128]  ; (0x2450c)
; OK, for the first argument of what we're calling, load the pointer at 2450c .. 03feb4.
00024500  e4 fe 03 08 d4 fe 03 08  c8 fe 03 08 b4 fe 03 08  |................|
; what's that a string pointer to
0003feb0  20 25 73 00 50 20 69 6e  74 65 72 76 61 6c 3a 25  | %s.P interval:%|
0003fec0  64 2f 25 64 00 00 00 00  72 63 52 61 74 65 3a 25  |d/%d....rcRate:%|
; ah, P interval
Cleanflight:
        case 3:
            blackboxPrintfHeaderLine("P interval:%d/%d", blackboxConfig()->rate_num, blackboxConfig()->rate_denom);

Raceflight:
; woohoo, we're in the right place.
; now, get 2 8 bit ints and out of the config -- at offsets 1068 and 1069 -- and put them in the format
; string.
   2448a:       f891 242d       ldrb.w  r2, [r1, #1069] ; 0x42d
   2448e:       f891 142c       ldrb.w  r1, [r1, #1068] ; 0x42c
; and let's call the function at 34b34 with these arguments-- the P inteval format string, and these 2 numbers. (rate_num, rate_denom)
   24492:       f010 fb4f       bl      0x34b34
```

mlyle, 2016-09-22:
next, blackboxPrintfHeaderLine.  Will it be the same as cleanflight's implementation too?  Let's see.
```
cleanflight:
void blackboxPrintfHeaderLine(const char *fmt, ...)
{
    va_list va;

    blackboxWrite('H');
    blackboxWrite(' ');

    va_start(va, fmt);

    int written = blackboxPrintfv(fmt, va);

    va_end(va);

    blackboxWrite('\n');

    blackboxHeaderBudget -= written + 3;

raceflight:
; stack manipulation
   34b34:       b40f            push    {r0, r1, r2, r3}
   34b36:       b530            push    {r4, r5, lr}
   34b38:       b083            sub     sp, #12
   34b3a:       ac06            add     r4, sp, #24
; 72 is the letter 'H'
   34b3c:       2048            movs    r0, #72 ; 0x48
   34b3e:       f854 5b04       ldr.w   r5, [r4], #4
; call the function at 2fe80 (blackboxWrite)
   34b42:       f7fb f99d       bl      0x2fe80
; 32 is the character space (' ')
   34b46:       2020            movs    r0, #32
; call the function at 2fe80 with ' ' as a parameter.
   34b48:       f7fb f99a       bl      0x2fe80
; setup varargs, call 2ed24 (blakcboxPrintfv) with the passed in parameters
  34b4c:       462a            mov     r2, r5
   34b4e:       4623            mov     r3, r4
   34b50:       4909            ldr     r1, [pc, #36]   ; (0x34b78)
   34b52:       9401            str     r4, [sp, #4]
   34b54:       2000            movs    r0, #0
   34b56:       f7fa f8e5       bl      0x2ed24
   34b5a:       4604            mov     r4, r0
; character 10 is newline ('\n')
   34b5c:       200a            movs    r0, #10
; call blackboxWrite('\n')
   34b5e:       f7fb f98f       bl      0x2fe80
; load blackboxHeaderBudget into r3
   34b62:       4a06            ldr     r2, [pc, #24]   ; (0x34b7c)
   34b64:       6813            ldr     r3, [r2, #0]
; add 3 to the value returned by blackboxPrintfv
   34b66:       3403            adds    r4, #3
; subtract this value from r3
   34b68:       1b1b            subs    r3, r3, r4
; and store the result back into blackboxHeaderBudget
   34b6a:       6013            str     r3, [r2, #0]
```

mlyle, 2016-09-22:
For clarity, I misnamed this issue.  I was going to reverse engineer altitude hold, but ... blackbox called to me so boldly in the string table.  Maybe altitude hold next :D

rs2k, 2016-09-22:
Let's keep the witch hunt down to one issue please.  :)

rs2k, 2016-09-22:
Duplicate of https://github.com/rs2k/raceflight/issues/225

closed 2016-09-22 by rs2k

Deckoz2302, 2016-09-22:
stop closing them when they have not been addressed :)