rs2k/raceflight #229: License violation - continued.

digitalentity, 2016-09-23:
As @rs2k didn't find any better way to be transparent than locking the #225, this is the new issue to discuss the matter.

tracernz, 2016-09-23:
If you really want to discuss, better to make the thread somewhere RS2K/proggod can't censor everything. :wink:

seaweeduk, 2016-09-23:
Is there an uncensored discussion on this issue going on anywhere? 

I think it's something the wider community should be made aware of. Seeing the censorship in the original issue, especially the comments from Boris which were deleted is extremely alarming to me.

digitalentity, 2016-09-23:
Should we open a thread on RCGroups?

seaweeduk, 2016-09-23:
I'm surprised if there isn't a thread there already, someone should make one for sure. Sounds like RaceFlight appear to be censoring discussion on this in their own thread with mentions of "keeping this thread clean of troll posts". I guess the comments they deleted from Boris were "troll posts" too right? I think someone needs to explain to them what the definition of a troll is, as well as what a GPL license is.